rmurai0610/MASt3R-SLAM #1: pip install -e thirdparty/mast3r  ERROR

ZTG0421, 2025-02-25:
ERROR: Could not find a version that satisfies the requirement faiss-gpu-cu12 (from asmk) (from versions: none)
ERROR: No matching distribution found for faiss-gpu-cu12

rmurai0610, 2025-02-25:
Hi, does `pip install faiss-gpu-cu12` work?
We've only tested our work on Ubuntu, what system are you using?

ZTG0421, 2025-02-25:
> 你好，`pip install faiss-gpu-cu12`有用吗？ 我们只在 Ubuntu 上测试过我们的工作，你使用的是什么系统？

ah , I use windows11

rmurai0610, 2025-02-25:
I see, unfortunately I'm not sure if this code will work on Windows. 
Please do let us know if you make it work though!

ZakeyShi, 2025-02-26:
hi @ZTG0421 @rmurai0610 , I think `faiss-gpu` is not support on Windows/MacOS. Currently the **GPU** version of `faiss` package is only available on Linux, you should install **CPU** version instead. ([Ref.](https://github.com/facebookresearch/faiss/blob/main/INSTALL.md))

Milesssssss, 2025-02-26:
@rmurai0610 If you are not sure, you still need to describe the applicable operating environment in the readme

rmurai0610, 2025-02-26:
@ZakeyShi Thanks, I've just remove the faiss-gpu requirement from pyproject as we weren't using it

closed 2025-02-26 by rmurai0610