rmurai0610/MASt3R-SLAM #3: Submodules not cloning

ricshaw, 2025-02-25:
Cloning of the repo is not working for me:

Cloning into './MASt3R-SLAM/thirdparty/in3d/thirdparty/pyimgui'...
ssh: connect to host github.com port 22: Connection timed out
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:pyimgui/pyimgui.git' into submodule path './MASt3R-SLAM/thirdparty/in3d/thirdparty/pyimgui' failed
Failed to clone 'thirdparty/in3d/thirdparty/pyimgui'. Retry scheduled

rmurai0610, 2025-02-25:
Hi can you try again? I've just merged the PR https://github.com/rmurai0610/MASt3R-SLAM/pull/2 which should solve this.

ricshaw, 2025-02-25:
That works, thanks!