rmurai0610/MASt3R-SLAM #4: Installation issues

ricshaw, 2025-02-25:
Hi, I am having some issues getting the code to run. I'm trying to run the command:
python main.py --dataset video.mp4 --config config/base.yaml
but encountering the following problems.

1) Seem to be having a glfw display error. Is it possible to run in headless mode? 
/home/mast3r-slam/lib/python3.11/site-packages/glfw/__init__.py:917: GLFWError: (65550) b'X11: The DISPLAY environment variable is missing'
  warnings.warn(message, GLFWError)
Process Process-2:
Traceback (most recent call last):
/home/mast3r-slam/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/miniconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "./MASt3R-SLAM/mast3r_slam/visualization.py", line 390, in run_visualization
    window = window_cls(
             ^^^^^^^^^^^
  File "/home/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/moderngl_window/context/glfw/window.py", line 31, in __init__
    raise ValueError("Failed to initialize glfw")
ValueError: Failed to initialize glfw

2) Some CUDA error:
  File "./MASt3R-SLAM/main.py", line 263, in <module>
    add_new_kf, match_info, try_reloc = tracker.track(frame)
                                        ^^^^^^^^^^^^^^^^^^^^
  File "./MASt3R-SLAM/mast3r_slam/tracker.py", line 31, in track
    idx_f2k, valid_match_k, Xff, Cff, Qff, Xkf, Ckf, Qkf = mast3r_match_asymmetric(
                                                           ^^^^^^^^^^^^^^^^^^^^^^^^
  File ./MASt3R-SLAM/mast3r_slam/mast3r_utils.py", line 221, in mast3r_match_asymmetric
    idx_i2j, valid_match_j = matching.match(
                             ^^^^^^^^^^^^^^^
  File "./MASt3R-SLAM/mast3r_slam/matching.py", line 9, in match
    idx_1_to_2, valid_match2 = match_iterative_proj(X11, X21, D11, D21, idx_1_to_2_init)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "./MASt3R-SLAM/mast3r_slam/matching.py", line 68, in match_iterative_proj
    p1 = p1.long()
         ^^^^^^^^^
RuntimeError: CUDA error: no kernel image is available for execution on the device
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

saeejithnair, 2025-02-26:
I had the same issue for a headless display, you can get around it by setting `--no-viz` flag in `main.py`

ricshaw, 2025-02-26:
Thanks, the --no-viz flag helped. But still getting this:

RuntimeError: CUDA error: no kernel image is available for execution on the device
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

Have tried using the three cuda versions in the instructions...

edexheim, 2025-02-26:
Which GPU architecture are you using?

ricshaw, 2025-02-26:
I’m using an Nvidia Tesla V100 GPU (Volta architecture, CUDA compute capability 7.0)

edexheim, 2025-02-26:
Thank you for the response. 

In this line, https://github.com/rmurai0610/MASt3R-SLAM/blob/76e68e4cc3a50f344193de5b80c10da419643d24/setup.py#L31 can you change the 86 to 70 in both places and then recompile?

If that works we'll include additional architectures which may slow compilation time but should hopefully work out of the box better.

youngkyoonjang, 2025-02-26:
It works! Thanks @edexheim and @ricshaw

KwanWaiPang, 2025-02-26:
> I had the same problem. After following [@edexheim](https://github.com/edexheim) 's feedback, It works on my GPU, NVIDIA GeForce RTX 2080! Thanks [@edexheim](https://github.com/edexheim) and [@ricshaw](https://github.com/ricshaw)

Thanks, I also fixed this problem.

BTW, as for visualization, do you meet the error like following:

~~~
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
~~~

[image]

edexheim, 2025-02-26:
Hi @KwanWaiPang ,

Can you make the libGL error a new issue? 

Thanks.

KwanWaiPang, 2025-02-26:
> Hi [@KwanWaiPang](https://github.com/KwanWaiPang) ,
> 
> Can you make the libGL error a new issue?
> 
> Thanks.

Problem solved through `ln -s /lib/x86_64-linux-gnu/dri/swrast_dri.so /usr/lib/dri`

ricshaw, 2025-02-26:
> Hi [@ricshaw](https://github.com/ricshaw),
> 
> We have added support for other architectures in [1cd221b](https://github.com/rmurai0610/MASt3R-SLAM/commit/1cd221bef87f16d56f1dad3d5c66c037bd581a7d).
> 
> Let us know if this works!

Yes it's working, thank you!

closed 2025-02-26 by ricshaw