rmurai0610/MASt3R-SLAM #5: Relocalization on Recorded Map and Resolution

glvov-bdai, 2025-02-25:
Thanks for releasing this repo! I was able to run the example on a realsense with impressive results.

Two quick questions:
- Is there a method to record a map and then perform camera relocalization on the known map?
- Is it possible to decrease the resolution of the realsense when running the live example?

glvov-bdai, 2025-02-25:
Found resolution at https://github.com/rmurai0610/MASt3R-SLAM/blob/76e68e4cc3a50f344193de5b80c10da419643d24/mast3r_slam/dataloader.py#L154

rmurai0610, 2025-02-26:
Yes, you can change the resolution there.
I've just pushed the ply export functionality. Please follow #7 to try!

glvov-bdai, 2025-02-26:
Hi @rmurai0610 thanks for a quick response!

One more question - is it possible to add intrinsics to the realsense demo?

When I modified the realsense, 

```
class RealsenseDataset(MonocularDataset):
    def __init__(self):
        super().__init__()
        self.use_calibration = True
        self.dataset_path = None
        self.pipeline = rs.pipeline()
        self.h, self.w = 240,320
        self.rs_config = rs.config()
        self.rs_config.enable_stream(
            rs.stream.color, self.w, self.h, rs.format.bgr8, 30
        )
        self.profile = self.pipeline.start(self.rs_config)

        self.rgb_sensor = self.profile.get_device().query_sensors()[1]
        # self.rgb_sensor.set_option(rs.option.enable_auto_exposure, False)
        # self.rgb_sensor.set_option(rs.option.enable_auto_white_balance, False)
        # self.rgb_sensor.set_option(rs.option.exposure, 200)
        self.rgb_profile = rs.video_stream_profile(
            self.profile.get_stream(rs.stream.color)
        )
        self.save_results = False
        fx = 426.7257080078125 #(first element)
        fy = 423.1294250488281 #(third element)
        cx = 426.7257080078125 #(fifth element)
        cy = 237.41268920898438 #(sixth element)
        self.camera_intrinsics = Intrinsics.from_calib(
            224, self.w, self.h, [fx, fy, cx, cy]
        )
````

and ran 
```
python main.py --dataset realsense --config config/calib.yaml
```

The script ran without apparent issues, but the viewer was empty except for the frustum.

This could totally be a skill issue but was wondering if you're able to replicate the issue

glvov-bdai, 2025-02-26:
Also one more comment; if you have a stored ply, how would you suggest running relocalization?

rmurai0610, 2025-02-26:
Relocalising against exisitng map is tricky to implement, and is not supported.
You'll need to export all of the state, and reload it before starting.

glvov-bdai, 2025-02-26:
Awesome thank you @rmurai0610 . Relocalization is on my wishlist ;)

finnBsch, 2025-05-30:
Following up on this:
Is the relocalization "just" an implementation feature that's missing, or are there conceptual problems in getting this to work? 
Really interested in the feature, too!