rmurai0610/MASt3R-SLAM #11: pip install --no-build-isolation  -e .  building error of MAST3R-SLAM

ppsohandsome, 2025-02-26:
this is the error message:

```
gn_kernels.obj : error LNK2001: 无法解析的外部符号 "public: long * __cdecl at::TensorBase::mutable_data_ptr<long>(void)const " (??$mutable_data_ptr@J@TensorBase@at@@QEBAPEAJXZ) 

matching_kernels.obj : error LNK2001: 无法解析的外部符号 "public: long * __cdecl at::TensorBase::mutable_data_ptr<long>(void)const " (??$mutable_data_ptr@J@TensorBase@at@@QEBAPEAJXZ)

build\lib.win-amd64-cpython-310\mast3r_slam_backends.cp310-win_amd64.pyd : fatal error LNK1120: 1 个无法解析的外部命令

error: command 'C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.43.34808\\bin\\HostX86\\x64\\link.exe' failed with exit code 1120
```

environment:
win 11
mscv 2022
pytorch 2.4.1+cu121

andrelevi, 2025-02-26:
So you finally got it working Windows? Just wondering if it's possible and worth the effort to try this on Windows. Thanks for following up with your solution

ppsohandsome, 2025-02-27:
directly windows，and not work yet. There is a package called torchcodec, when I search the index of torchcodec, no version can be used in windows. 

I don't know whether I miss something, if you finish it,please let me know.

godisme1220, 2025-02-27:
Hi, after I search the whole internet and ask the torchcodec dev team, unfortunately there is no Windows version of torchcodec, only supports Linux. I just raise an issue in torchcodec github page, [see here](https://github.com/pytorch/torchcodec/issues/522).

So currently seems not support Windows platform now, but what exactly torchcodec do in this project? Maybe can use some alternate package?

Best regrads Eugene.

aeaeaeaeaeae, 2025-02-27:
Does it run on WSL?

godisme1220, 2025-02-27:
> Does it run on WSL?

Yes, I successfully install on my Windows11 platform, with WSL2 Ubuntu-20.04.
I test both on cuda 11.8 and cuda 12.4 version, and successfully install all package, hope this informations help.

[image]

borokov, 2025-02-28:
> 
> Yes, I successfully install on my Windows11 platform, with WSL2 Ubuntu-20.04. I test both on cuda 11.8 and cuda 12.4 version, and successfully install all package, hope this informations help.
> 
> [image]

Does it also run ?
I'm on WSL2 Windows11, ubuntu-20.04, cuda 12.4. It builds, but crash at startup with:

[image]

godisme1220, 2025-02-28:
> Does it also run ? I'm on WSL2 Windows11, ubuntu-20.04, cuda 12.4. It builds, but crash at startup with:
>
> [image]

Yes! It did run on my WSL2, and the imgui UI did appear. Have you install all package correctly?

rmurai0610, 2025-02-28:
Hi, we've identified the issue for WSL #21. Hopefully, this works.

@ppsohandsome Thank you for the fix for building on visual studio. 
I'm not too sure if the code will work directly on Windows, but please let us know if anyone gets it working though!

closed 2025-03-03 by edexheim

suvodip1212, 2025-06-23:
It works on windows now. However, sometimes the GPU becomes unresponsive due to the curope backend. Still trying to fix that bug.