rmurai0610/MASt3R-SLAM #12: RuntimeError: CUDA error: no kernel image is available for execution on the device

KwanWaiPang, 2025-02-26:
hi~
First of all, I would like to thank you for your excellent work and for making the code open source.

While evaluating the Eurocdataset and running the following command:
~~~
python main.py --dataset datasets/euroc/MH_01_easy/ --no-viz --config config/eval_no_calib.yaml
~~~

I encountered the following error:
~~~
  File "/home/gwp/MASt3R-SLAM/main.py", line 263, in <module>
    add_new_kf, match_info, try_reloc = tracker.track(frame)
                                        ^^^^^^^^^^^^^^^^^^^^
  File "/home/gwp/MASt3R-SLAM/mast3r_slam/tracker.py", line 31, in track
    idx_f2k, valid_match_k, Xff, Cff, Qff, Xkf, Ckf, Qkf = mast3r_match_asymmetric(
                                                           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gwp/MASt3R-SLAM/mast3r_slam/mast3r_utils.py", line 221, in mast3r_match_asymmetric
    idx_i2j, valid_match_j = matching.match(
                             ^^^^^^^^^^^^^^^
  File "/home/gwp/MASt3R-SLAM/mast3r_slam/matching.py", line 9, in match
    idx_1_to_2, valid_match2 = match_iterative_proj(X11, X21, D11, D21, idx_1_to_2_init)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gwp/MASt3R-SLAM/mast3r_slam/matching.py", line 68, in match_iterative_proj
    p1 = p1.long()
         ^^^^^^^^^
RuntimeError: CUDA error: no kernel image is available for execution on the device
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.
~~~

[image]

Any idea for this?

My CUDA is 12.2 and I install pytorch as following:
~~~
conda install pytorch==2.5.1 torchvision==0.20.1 torchaudio==2.5.1 pytorch-cuda=12.1 -c pytorch -c nvidia
~~~

Thank you for your help, and looking forward to your response.

LeiyuanMa, 2025-02-26:
same question

edexheim, 2025-02-26:
Yes, we have now updated the code to handle more GPU architectures in 1cd221bef87f16d56f1dad3d5c66c037bd581a7d.

closed 2025-02-26 by edexheim