rmurai0610/MASt3R-SLAM #13: ERROR: Failed building wheel for lietorch on WSL2 Ubuntu 20.04 (CUDA 12.4)

godisme1220, 2025-02-26:
I encountered an error when running the installation command:
pip install --no-build-isolation -e .

And end up with this error, how can i solve it?

  Building wheel for lietorch (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [3942 lines of output]
.....
.....
.....
      RuntimeError: Error compiling objects for extension
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for lietorch
  Running setup.py clean for lietorch
Successfully built MAST3R-SLAM
Failed to build lietorch
ERROR: Failed to build installable wheels for some pyproject.toml based projects (lietorch)

edexheim, 2025-02-26:
Can you try following the [lietorch](https://github.com/princeton-vl/lietorch) instructions to install and see if there are any specific errors raised?

Thanks,
Eric

godisme1220, 2025-02-27:
> Can you try following the [lietorch](https://github.com/princeton-vl/lietorch) instructions to install and see if there are any specific errors raised?
> 
> Thanks, Eric

Dear Eric thanks for reply! After I follow the lietorch instructions, it did install correctly and the error keep happening was due to the OOM(out of memory) error problem, because I'm running in WSL2 Ubuntu20.04 system, default memory limit is set to 8GB, after I adjust to 12GB, the lietorch package did install correctly without any errors.

Eugene best regards.

edexheim, 2025-02-27:
Thanks! Glad it worked.