rmurai0610/MASt3R-SLAM #17: OSError: libEGL.so

ZTG0421, 2025-02-26:
(mast3r-slam) ztg@ZTGU:~/Desktop/MASt3R-SLAM$ python main.py --dataset datasets/tum/rgbd_dataset_freiburg1_room/ --config config/calib.yaml
config/calib.yaml
config/base.yaml
datasets/tum/rgbd_dataset_freiburg1_room/
{'use_calib': True, 'single_thread': False, 'dataset': {'subsample': 2, 'img_downsample': 1, 'center_principle_point': True}, 'matching': {'max_iter': 10, 'lambda_init': 1e-08, 'convergence_thresh': 1e-06, 'dist_thresh': 0.1, 'radius': 3, 'dilation_max': 5}, 'tracking': {'min_match_frac': 0.05, 'max_iters': 50, 'C_conf': 0.0, 'Q_conf': 1.5, 'rel_error': 0.001, 'delta_norm': 0.001, 'huber': 1.345, 'match_frac_thresh': 0.333, 'sigma_ray': 0.003, 'sigma_dist': 10.0, 'sigma_pixel': 1.0, 'sigma_depth': 10.0, 'sigma_point': 0.05, 'pixel_border': -10, 'depth_eps': 1e-06, 'filtering_mode': 'weighted_pointmap', 'filtering_score': 'median'}, 'local_opt': {'pin': 1, 'window_size': 1000000.0, 'C_conf': 0.0, 'Q_conf': 1.5, 'min_match_frac': 0.1, 'pixel_border': -10, 'depth_eps': 1e-06, 'max_iters': 10, 'sigma_ray': 0.003, 'sigma_dist': 10.0, 'sigma_pixel': 1.0, 'sigma_depth': 10.0, 'sigma_point': 0.05, 'delta_norm': 1e-08, 'use_cuda': True}, 'retrieval': {'k': 3, 'min_thresh': 0.005}, 'reloc': {'min_match_frac': 0.3, 'strict': True}, 'inherit': 'config/base.yaml'}
... loading model from checkpoints/MASt3R_ViTLarge_BaseDecoder_512_catmlpdpt_metric.pth
/home/ztg/Desktop/MASt3R-SLAM/thirdparty/mast3r/mast3r/model.py:24: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  ckpt = torch.load(model_path, map_location='cpu')
instantiating : AsymmetricMASt3R(enc_depth=24, dec_depth=12, enc_embed_dim=1024, dec_embed_dim=768, enc_num_heads=16, dec_num_heads=12, pos_embed='RoPE100',img_size=(512, 512), head_type='catmlp+dpt', output_mode='pts3d+desc24', depth_mode=('exp', -inf, inf), conf_mode=('exp', 1, inf), patch_embed_cls='PatchEmbedDust3R', two_confs=True, desc_conf_mode=('exp', 0, inf), landscape_only=False)
config/calib.yaml
config/base.yaml
Process Process-2:
Traceback (most recent call last):
  File "/home/ztg/anaconda3/envs/mast3r-slam/lib/python3.11/site-packages/_moderngl.py", line 214, in __init__
    lib = ctypes.CDLL("libEGL.so")
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ztg/anaconda3/envs/mast3r-slam/lib/python3.11/ctypes/__init__.py", line 376, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: libEGL.so: 无法打开共享目标文件: 没有那个文件或目录

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ztg/anaconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/ztg/anaconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/ztg/Desktop/MASt3R-SLAM/mast3r_slam/visualization.py", line 390, in run_visualization
    window = window_cls(
             ^^^^^^^^^^^
  File "/home/ztg/anaconda3/envs/mast3r-slam/lib/python3.11/site-packages/moderngl_window/context/glfw/window.py", line 85, in __init__
    self.init_mgl_context()
  File "/home/ztg/anaconda3/envs/mast3r-slam/lib/python3.11/site-packages/moderngl_window/context/base/window.py", line 173, in init_mgl_context
    self._ctx = moderngl.create_context(require=self.gl_version_code)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ztg/anaconda3/envs/mast3r-slam/lib/python3.11/site-packages/moderngl/__init__.py", line 2242, in create_context
    ctx = get_context()
          ^^^^^^^^^^^^^
  File "/home/ztg/anaconda3/envs/mast3r-slam/lib/python3.11/site-packages/moderngl/__init__.py", line 2299, in get_context
    init_context()
  File "/home/ztg/anaconda3/envs/mast3r-slam/lib/python3.11/site-packages/moderngl/__init__.py", line 2281, in init_context
    loader = DefaultLoader()
             ^^^^^^^^^^^^^^^
  File "/home/ztg/anaconda3/envs/mast3r-slam/lib/python3.11/site-packages/_moderngl.py", line 223, in __init__
    lib = ctypes.CDLL("libGL.so")
          ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ztg/anaconda3/envs/mast3r-slam/lib/python3.11/ctypes/__init__.py", line 376, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: libGL.so: 无法打开共享目标文件: 没有那个文件或目录
^CTraceback (most recent call last):
  File "/home/ztg/Desktop/MASt3R-SLAM/main.py", line 182, in <module>
    model = load_mast3r(device=device)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ztg/Desktop/MASt3R-SLAM/mast3r_slam/mast3r_utils.py", line 20, in load_mast3r
    model = AsymmetricMASt3R.from_pretrained(weights_path).to(device)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ztg/Desktop/MASt3R-SLAM/thirdparty/mast3r/mast3r/model.py", line 50, in from_pretrained
    return load_model(pretrained_model_name_or_path, device='cpu')
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ztg/Desktop/MASt3R-SLAM/thirdparty/mast3r/mast3r/model.py", line 33, in load_model
    net = eval(args)
          ^^^^^^^^^^
  File "<string>", line 1, in <module>
  File "/home/ztg/Desktop/MASt3R-SLAM/thirdparty/mast3r/mast3r/model.py", line 45, in __init__
    super().__init__(**kwargs)
  File "/home/ztg/Desktop/MASt3R-SLAM/thirdparty/mast3r/dust3r/dust3r/model.py", line 73, in __init__
    self.set_downstream_head(output_mode, head_type, landscape_only, depth_mode, conf_mode, **croco_kwargs)
  File "/home/ztg/Desktop/MASt3R-SLAM/thirdparty/mast3r/mast3r/model.py", line 64, in set_downstream_head
    self.downstream_head1 = mast3r_head_factory(head_type, output_mode, self, has_conf=bool(conf_mode))
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ztg/Desktop/MASt3R-SLAM/thirdparty/mast3r/mast3r/catmlp_dpt_head.py", line 111, in mast3r_head_factory
    return Cat_MLP_LocalFeatures_DPT_Pts3d(net, local_feat_dim=local_feat_dim, has_conf=has_conf,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ztg/Desktop/MASt3R-SLAM/thirdparty/mast3r/mast3r/catmlp_dpt_head.py", line 67, in __init__
    self.head_local_features = Mlp(in_features=idim,
                               ^^^^^^^^^^^^^^^^^^^^^
  File "/home/ztg/Desktop/MASt3R-SLAM/thirdparty/mast3r/dust3r/croco/models/blocks.py", line 70, in __init__
    self.fc2 = nn.Linear(hidden_features, out_features, bias=bias[1])
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ztg/anaconda3/envs/mast3r-slam/lib/python3.11/site-packages/torch/nn/modules/linear.py", line 112, in __init__
    self.reset_parameters()
  File "/home/ztg/anaconda3/envs/mast3r-slam/lib/python3.11/site-packages/torch/nn/modules/linear.py", line 118, in reset_parameters
    init.kaiming_uniform_(self.weight, a=math.sqrt(5))
  File "/home/ztg/anaconda3/envs/mast3r-slam/lib/python3.11/site-packages/torch/nn/init.py", line 518, in kaiming_uniform_
    return tensor.uniform_(-bound, bound, generator=generator)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt

ZTG0421, 2025-02-26:
HI,I just install ubuntu22 ,then follow README.

rmurai0610, 2025-02-26:
Hi are you running it on a server? Or a desktop/laptop?

ZTG0421, 2025-02-27:
> Hi are you running it on a server? Or a desktop/laptop?

ubuntu desktop on my pc

godisme1220, 2025-02-27:
Im running on Windows11 with WSL2 Ubuntu 20.04, and got similar problem.

Terminal Output:
(mast3r-slam) devcontainers@EugeneLin-2024:/mnt/c/python_project/WSL/MASt3R-SLAM$ python main.py --dataset /mnt/c/python_project/WSL/MASt3R-SLAM/datasets/taichung_explode.mp4 --config config/base.yaml
config/base.yaml
/mnt/c/python_project/WSL/MASt3R-SLAM/datasets/taichung_explode.mp4
{'use_calib': False, 'single_thread': False, 'dataset': {'subsample': 1, 'img_downsample': 1, 'center_principle_point': True}, 'matching': {'max_iter': 10, 'lambda_init': 1e-08, 'convergence_thresh': 1e-06, 'dist_thresh': 0.1, 'radius': 3, 'dilation_max': 5}, 'tracking': {'min_match_frac': 0.05, 'max_iters': 50, 'C_conf': 0.0, 'Q_conf': 1.5, 'rel_error': 0.001, 'delta_norm': 0.001, 'huber': 1.345, 'match_frac_thresh': 0.333, 'sigma_ray': 0.003, 'sigma_dist': 10.0, 'sigma_pixel': 1.0, 'sigma_depth': 10.0, 'sigma_point': 0.05, 'pixel_border': -10, 'depth_eps': 1e-06, 'filtering_mode': 'weighted_pointmap', 'filtering_score': 'median'}, 'local_opt': {'pin': 1, 'window_size': 1000000.0, 'C_conf': 0.0, 'Q_conf': 1.5, 'min_match_frac': 0.1, 'pixel_border': -10, 'depth_eps': 1e-06, 'max_iters': 10, 'sigma_ray': 0.003, 'sigma_dist': 10.0, 'sigma_pixel': 1.0, 'sigma_depth': 10.0, 'sigma_point': 0.05, 'delta_norm': 1e-08, 'use_cuda': True}, 'retrieval': {'k': 3, 'min_thresh': 0.005}, 'reloc': {'min_match_frac': 0.3, 'strict': True}}
... loading model from checkpoints/MASt3R_ViTLarge_BaseDecoder_512_catmlpdpt_metric.pth
/mnt/c/python_project/WSL/MASt3R-SLAM/thirdparty/mast3r/mast3r/model.py:24: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  ckpt = torch.load(model_path, map_location='cpu')
config/base.yaml
Process Process-2:
Traceback (most recent call last):
  File "/home/devcontainers/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/_moderngl.py", line 214, in __init__
    lib = ctypes.CDLL("libEGL.so")
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/devcontainers/miniconda3/envs/mast3r-slam/lib/python3.11/ctypes/__init__.py", line 376, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: libEGL.so: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/devcontainers/miniconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/devcontainers/miniconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/mnt/c/python_project/WSL/MASt3R-SLAM/mast3r_slam/visualization.py", line 390, in run_visualization
    window = window_cls(
             ^^^^^^^^^^^
  File "/home/devcontainers/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/moderngl_window/context/glfw/window.py", line 85, in __init__
    self.init_mgl_context()
  File "/home/devcontainers/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/moderngl_window/context/base/window.py", line 173, in init_mgl_context
    self._ctx = moderngl.create_context(require=self.gl_version_code)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/devcontainers/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/moderngl/__init__.py", line 2242, in create_context
    ctx = get_context()
          ^^^^^^^^^^^^^
  File "/home/devcontainers/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/moderngl/__init__.py", line 2299, in get_context
    init_context()
  File "/home/devcontainers/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/moderngl/__init__.py", line 2281, in init_context
    loader = DefaultLoader()
             ^^^^^^^^^^^^^^^
  File "/home/devcontainers/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/_moderngl.py", line 223, in __init__
    lib = ctypes.CDLL("libGL.so")
          ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/devcontainers/miniconda3/envs/mast3r-slam/lib/python3.11/ctypes/__init__.py", line 376, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: libGL.so: cannot open shared object file: No such file or directory

All package were installed correctly, but when I input video command, error happended.
"python main.py --dataset /mnt/c/python_project/WSL/MASt3R-SLAM/datasets/taichung_explode.mp4 --config config/base.yaml"

First the new window will pop up, but terminate quickly, then error output, but the process still going, eventually print out done.
FPS: 7.249249388430781
FPS: 7.249319545479217
FPS: 7.248996230255473
FPS: 7.248343540795231
FPS: 7.246894025728771
FPS: 7.24542636559673
FPS: 7.2453971752826964
FPS: 7.245511395845407
FPS: 7.245719304705426
FPS: 7.245530610651535
FPS: 7.245693981056713
done

But when I open ply file in blender, all i got is a lot of point in origin position, seems not correct.

godisme1220, 2025-02-27:
I install these two package then it works correct.
```
sudo apt install libx11-dev
sudo apt install libgl1-mesa-dev
```

ZTG0421, 2025-02-27:
> 我安装了这两个包然后它运行正常。
> 
> ```
> sudo apt install libx11-dev
> sudo apt install libgl1-mesa-dev
> ```

Thanks, it works

closed 2025-02-27 by edexheim

youwyu, 2025-07-19:
The above alone did not work for me. I fixed it by installing
`conda install -c conda-forge libstdcxx-ng`