rmurai0610/MASt3R-SLAM #18: torch.OutOfMemoryError: CUDA out of memory.

ndor, 2025-02-26:
So, luckily, no much issues with installation.

However, try to run a video, and got the below error.
(same happened for images folder of the same video, original size and after, tried smaller sized images, of factor fx=fy=0.5).
Original video resolution - 1440 × 1080 pixels.

CUDA seems to overflow GRAM. I have a A200-Nvidia card (laptop-PC). Ubuntu 22.04 OS.

here is the printout:

```
config/base.yaml
/home/ndor/Desktop/freestyle_trials/PRODUCT_V0/images
{'use_calib': False, 'single_thread': False, 'dataset': {'subsample': 1, 'img_downsample': 1, 'center_principle_point': True}, 'matching': {'max_iter': 10, 'lambda_init': 1e-08, 'convergence_thresh': 1e-06, 'dist_thresh': 0.1, 'radius': 3, 'dilation_max': 5}, 'tracking': {'min_match_frac': 0.05, 'max_iters': 50, 'C_conf': 0.0, 'Q_conf': 1.5, 'rel_error': 0.001, 'delta_norm': 0.001, 'huber': 1.345, 'match_frac_thresh': 0.333, 'sigma_ray': 0.003, 'sigma_dist': 10.0, 'sigma_pixel': 1.0, 'sigma_depth': 10.0, 'sigma_point': 0.05, 'pixel_border': -10, 'depth_eps': 1e-06, 'filtering_mode': 'weighted_pointmap', 'filtering_score': 'median'}, 'local_opt': {'pin': 1, 'window_size': 1000000.0, 'C_conf': 0.0, 'Q_conf': 1.5, 'min_match_frac': 0.1, 'pixel_border': -10, 'depth_eps': 1e-06, 'max_iters': 10, 'sigma_ray': 0.003, 'sigma_dist': 10.0, 'sigma_pixel': 1.0, 'sigma_depth': 10.0, 'sigma_point': 0.05, 'delta_norm': 1e-08, 'use_cuda': True}, 'retrieval': {'k': 3, 'min_thresh': 0.005}, 'reloc': {'min_match_frac': 0.3, 'strict': True}}
... loading model from checkpoints/MASt3R_ViTLarge_BaseDecoder_512_catmlpdpt_metric.pth
/home/ndor/Desktop/MASt3R-SLAM_trials/MASt3R-SLAM/thirdparty/mast3r/mast3r/model.py:24: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  ckpt = torch.load(model_path, map_location='cpu')
instantiating : AsymmetricMASt3R(enc_depth=24, dec_depth=12, enc_embed_dim=1024, dec_embed_dim=768, enc_num_heads=16, dec_num_heads=12, pos_embed='RoPE100',img_size=(512, 512), head_type='catmlp+dpt', output_mode='pts3d+desc24', depth_mode=('exp', -inf, inf), conf_mode=('exp', 1, inf), patch_embed_cls='PatchEmbedDust3R', two_confs=True, desc_conf_mode=('exp', 0, inf), landscape_only=False)
config/base.yaml
libGL error: MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
/home/ndor/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/glfw/__init__.py:917: GLFWError: (65543) b'GLX: Failed to create context: GLXBadFBConfig'
  warnings.warn(message, GLFWError)
Process Process-2:
Traceback (most recent call last):
  File "/home/ndor/miniconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/ndor/miniconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/ndor/Desktop/MASt3R-SLAM_trials/MASt3R-SLAM/mast3r_slam/visualization.py", line 390, in run_visualization
    window = window_cls(
             ^^^^^^^^^^^
  File "/home/ndor/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/moderngl_window/context/glfw/window.py", line 60, in __init__
    raise ValueError("Failed to create window")
ValueError: Failed to create window
<All keys matched successfully>
config/base.yaml
Loading retrieval model from checkpoints/MASt3R_ViTLarge_BaseDecoder_512_catmlpdpt_metric_retrieval_trainingfree.pth
/home/ndor/Desktop/MASt3R-SLAM_trials/MASt3R-SLAM/thirdparty/mast3r/mast3r/retrieval/processor.py:70: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  ckpt = torch.load(modelname, 'cpu')  # TODO from pretrained to download it automatically
/home/ndor/Desktop/MASt3R-SLAM_trials/MASt3R-SLAM/mast3r_slam/frame.py:116: UserWarning: The given NumPy array is not writable, and PyTorch does not support non-writable tensors. This means writing to this tensor will result in undefined behavior. You may want to copy the array to protect its data or make it writable before converting it to a tensor. This type of warning will be suppressed for the rest of this program. (Triggered internally at /opt/conda/conda-bld/pytorch_1729647429097/work/torch/csrc/utils/tensor_numpy.cpp:206.)
  uimg = torch.from_numpy(img["unnormalized_img"]) / 255.0
Traceback (most recent call last):
  File "/home/ndor/Desktop/MASt3R-SLAM_trials/MASt3R-SLAM/main.py", line 254, in <module>
    X_init, C_init = mast3r_inference_mono(model, frame)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ndor/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/torch/utils/_contextlib.py", line 116, in decorate_context
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/ndor/Desktop/MASt3R-SLAM_trials/MASt3R-SLAM/mast3r_slam/mast3r_utils.py", line 127, in mast3r_inference_mono
    res11, res21 = decoder(model, feat, feat, pos, pos, shape, shape)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ndor/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/torch/utils/_contextlib.py", line 116, in decorate_context
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/ndor/Desktop/MASt3R-SLAM_trials/MASt3R-SLAM/mast3r_slam/mast3r_utils.py", line 38, in decoder
    res1 = model._downstream_head(1, [tok.float() for tok in dec1], shape1)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ndor/Desktop/MASt3R-SLAM_trials/MASt3R-SLAM/thirdparty/mast3r/dust3r/dust3r/model.py", line 196, in _downstream_head
    return head(decout, img_shape)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ndor/Desktop/MASt3R-SLAM_trials/MASt3R-SLAM/thirdparty/mast3r/dust3r/dust3r/utils/misc.py", line 60, in transpose_to_landscape_no
    res = head(decout, (H, W))
          ^^^^^^^^^^^^^^^^^^^^
  File "/home/ndor/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/torch/nn/modules/module.py", line 1736, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
[... 43 lines cut ...]
    self._target(*self._args, **self._kwargs)
  File "/home/ndor/Desktop/MASt3R-SLAM_trials/MASt3R-SLAM/main.py", line 82, in run_backend
    mode = states.get_mode()
[... cut ...]
  File "/home/ndor/miniconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/connection.py", line 250, in recv
    buf = self._recv_bytes()
          ^^^^^^^^^^^^^^^^^^
  File "/home/ndor/miniconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/connection.py", line 430, in _recv_bytes
    buf = self._recv(4)
          ^^^^^^^^^^^^^
  File "/home/ndor/miniconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/connection.py", line 395, in _recv
    chunk = read(handle, remaining)
            ^^^^^^^^^^^^^^^^^^^^^^^
ConnectionResetError: [Errno 104] Connection reset by peer
```

thanks for any help!!

rmurai0610, 2025-02-26:
Hi, unfortunately, running MASt3R alone takes quite a lot of VRAM, and I am not too sure if we can run it on A200.
The best would be to run it without the viewer with `--no-viz` option but even then the VRAM might not be enough

ndor, 2025-02-26:
Thanks for the prompt reply @rmurai0610 !
What is the baseline GPU specs that can be used here? (A200 has 8GB of VRAM)
Is it only a VRAM issue?

BriannaBalamV, 2025-02-27:
> Thanks for the prompt reply [@rmurai0610](https://github.com/rmurai0610) ! What is the baseline GPU specs that can be used here? (A2000 has 8GB of VRAM) Is it only a VRAM issue?

Hi, did you solve the problem with /--no-viz?

I have just installed MASt3r-SLAM and I have this error while trying to execute the example:

<details><summary>Details</summary>
<p>

... loading model from checkpoints/MASt3R_ViTLarge_BaseDecoder_512_catmlpdpt_metric.pth
/home/brianna/MASt3R-SLAM/thirdparty/mast3r/mast3r/model.py:24: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  ckpt = torch.load(model_path, map_location='cpu')
instantiating : AsymmetricMASt3R(enc_depth=24, dec_depth=12, enc_embed_dim=1024, dec_embed_dim=768, enc_num_heads=16, dec_num_heads=12, pos_embed='RoPE100',img_size=(512, 512), head_type='catmlp+dpt', output_mode='pts3d+desc24', depth_mode=('exp', -inf, inf), conf_mode=('exp', 1, inf), patch_embed_cls='PatchEmbedDust3R', two_confs=True, desc_conf_mode=('exp', 0, inf), landscape_only=False)
config/TUM1.yaml
<All keys matched successfully>
Traceback (most recent call last):
  File "/home/brianna/MASt3R-SLAM/main.py", line 182, in <module>
    model = load_mast3r(device=device)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/brianna/MASt3R-SLAM/mast3r_slam/mast3r_utils.py", line 20, in load_mast3r
    model = AsymmetricMASt3R.from_pretrained(weights_path).to(device)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/brianna/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/torch/nn/modules/module.py", line 1340, in to
    return self._apply(convert)
           ^^^^^^^^^^^^^^^^^^^^
  File "/home/brianna/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/torch/nn/modules/module.py", line 900, in _apply
    module._apply(fn)
  File "/home/brianna/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/torch/nn/modules/module.py", line 900, in _apply
    module._apply(fn)
  File "/home/brianna/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/torch/nn/modules/module.py", line 900, in _apply
    module._apply(fn)
  [Previous line repeated 1 more time]
  File "/home/brianna/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/torch/nn/modules/module.py", line 927, in _apply
    param_applied = fn(param)
                    ^^^^^^^^^
  File "/home/brianna/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/torch/nn/modules/module.py", line 1326, in convert
    return t.to(
           ^^^^^
torch.OutOfMemoryError: CUDA out of memory. Tried to allocate 20.00 MiB. GPU 0 has a total capacity of 5.68 GiB of which 4.56 MiB is free. Including non-PyTorch memory, this process has 5.51 GiB memory in use. Process 56566 has 126.00 MiB memory in use. Of the allocated memory 5.38 GiB is allocated by PyTorch, and 13.78 MiB is reserved by PyTorch but unallocated. If reserved but unallocated memory is large try setting PYTORCH_CUDA_ALLOC_CONF=expandable_segments:True to avoid fragmentation.  See documentation for Memory Management  (https://pytorch.org/docs/stable/notes/cuda.html#environment-variables)
Process Process-2:
Traceback (most recent call last):
  File "/home/brianna/miniconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/brianna/miniconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/brianna/MASt3R-SLAM/mast3r_slam/visualization.py", line 438, in run_visualization
    window.render(current_time, delta)
  File "/home/brianna/miniconda3/envs/mast3r-slam/lib/python3.11/site-packages/moderngl_window/context/base/window.py", line 742, in render
    self.render_func(time, frame_time)
  File "/home/brianna/MASt3R-SLAM/mast3r_slam/visualization.py", line 131, in render
    with self.keyframes.lock:
  File "/home/brianna/miniconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/managers.py", line 1056, in __enter__
    return self._callmethod('acquire')
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/brianna/miniconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/managers.py", line 837, in _callmethod
    raise convert_to_error(kind, result)
multiprocessing.managers.RemoteError: 
---------------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/brianna/miniconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/managers.py", line 260, in serve_client
    self.id_to_local_proxy_obj[ident]
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^
KeyError: '7fc384255180'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/brianna/miniconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/managers.py", line 262, in serve_client
    raise ke
  File "/home/brianna/miniconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/managers.py", line 256, in serve_client
    obj, exposed, gettypeid = id_to_obj[ident]
                              ~~~~~~~~~^^^^^^^
KeyError: '7fc384255180'
---------------------------------------------------------------------------

</p>
</details>

ndor, 2025-02-27:
> --no-viz

same, unfortunately

rmurai0610, 2025-02-27:
I'm not too sure if this will work, but you can try and set https://github.com/rmurai0610/MASt3R-SLAM/blob/main/main.py#L172
to
`keyframes = SharedKeyframes(manager, h, w, 64)`
This reduce the overall number of allocated keyframes (or set it lower).
But this will fail for longer sequence which requires over 64 keyframes.

BriannaBalamV, 2025-02-27:
Hi, thanks that worked!!

[image]

closed 2025-02-28 by rmurai0610

ppsohandsome, 2025-03-01:
when i change` h, w = dataset.get_img_shape()[0]`into`keyframes = SharedKeyframes(manager, h, w, 64)` ,there is a error ? if i change the wrong line?

rmurai0610, 2025-03-01:
Hi, after new commits the lines changed but you need to change
`
keyframes = SharedKeyframes(manager, h, w)
`
To
`
keyframes = SharedKeyframes(manager, h, w, 64)`

Kishlay-notabot, 2025-03-21:
Does this only work on CUDA?

suayu, 2025-05-15:
Thanks.Your method is very useful. My NVIDIA GeForce RTX 3070 with 8G mem can run the demo.

> I'm not too sure if this will work, but you can try and set https://github.com/rmurai0610/MASt3R-SLAM/blob/main/main.py#L172 to `keyframes = SharedKeyframes(manager, h, w, 64)` This reduce the [...]