rmurai0610/MASt3R-SLAM #9: [Bug] compile error when building mast3r-slam package

ZakeyShi, 2025-02-26:
### Description
I'm following the instructions for installation in the readme, and I get an error when I get to 
```bash
pip install --no-build-isolation -e .
```
and all the previous instructions execute successfully.

### Environment

- System: ubuntu-20.04 
- nvcc version: 11.7
- torch version: 2.0.1
> I check my GPU driver version using `nvidia-smi` and `nvcc -V`, and install pytorch using command:` conda install pytorch==2.0.1 torchvision==0.15.2 torchaudio==2.0.2 pytorch-cuda=11.7 -c pytorch -c nvidia` followed [here](https://pytorch.org/get-started/previous-versions/)

### Error info

```bash
.../mast3r-slam/mast3r_slam/backend/src/gn_kernels.cu(802): error: no instance of overloaded function "torch::linalg::linalg_norm" matches the argument list
              argument types are: (at::Tensor, std::optional<c10::Scalar>, {...}, __nv_bool, {...})
```

### Solution

The reason for this problem comes from an interface change in [pytorch-2.5.0](https://github.com/pytorch/pytorch/commit/ff7b6d6b5fb343c7e5e72c8a9ee4df3302396867).

For pytorch < 2.5.0, `linalg_norm` function signature is [[Ref.](https://github.com/pytorch/pytorch/blob/v2.4.0/aten/src/ATen/native/LinearAlgebra.cpp#L3004)]
``` bash
Tensor linalg_norm(const Tensor& X, const optional<Scalar>& opt_ord, OptionalIntArrayRef opt_dim, bool keepdim, std::optional<ScalarType> opt_dtype) 
```

But for pytorch >= 2.5.0, `linalg_norm` function signature is [[Ref](https://github.com/pytorch/pytorch/blob/v2.5.0/aten/src/ATen/native/LinearAlgebra.cpp#L2995).]
``` bash
Tensor linalg_norm(const Tensor& X, const std::optional<Scalar>& opt_ord, OptionalIntArrayRef opt_dim, bool keepdim, std::optional<ScalarType> opt_dtype) 
```

Obviously, the second argument of the function is of a different type. So for pytorch<2.5.0, the following code needs to be modified from

https://github.com/rmurai0610/MASt3R-SLAM/blob/76e68e4cc3a50f344193de5b80c10da419643d24/mast3r_slam/backend/src/gn_kernels.cu#L802

to

```bash
delta_norm = torch::linalg::linalg_norm(dx, c10::optional<c10::Scalar>(), {}, false, {});
```

The other `torch::linalg::linalg_norm` function calls in the [gu_kernels.cu](mast3r_slam/backend/src/gn_kernels.cu) file are modified in the same way.

Then `pip install --no-build-isolation -e .`  can run successfully！

xiaohulihutu, 2025-02-26:
works! thats the issue. thanks!

bnkf1156f, 2025-03-15:
further if you are using cuda=11.7 and pytorch=2.0.1, it might cause problem during inference so:
change each
`@torch.inference_mode`
to
`@torch.inference_mode()`
in  [mast3r_slam/mast3r_utils.py](https://github.com/rmurai0610/MASt3R-SLAM/blob/main/mast3r_slam/mast3r_utils.py)