rmurai0610/MASt3R-SLAM #21: No Reconstruction Appear.

godisme1220, 2025-02-27:
Dear team, thanks for this amazing project.
I successfully install the enviroment, and also try the video process command, the command i use:

`python main.py --dataset /mnt/c/python_project/WSL/MASt3R-SLAM/datasets/IMG_0699.mp4 --config config/base.yaml`

When I execute the command, the imgui window appear, and video is playing. But different from the main page gif image I saw, the upper image stopped at frame 1 and terminal print Skipped keyframe1, then after I wait the process done, there is still nothing reconstuction appears, seems not correct, and I look in the log folder, open txt and ply file, nothing shows, only bunch of zero data. Is this normal? How can I fix it?

Terminal Output:
`/mnt/c/python_project/WSL/MASt3R-SLAM/datasets/IMG_0699.mp4
{'use_calib': False, 'single_thread': False, 'dataset': {'subsample': 1, 'img_downsample': 1, 'center_principle_point': True}, 'matching': {'max_iter': 10, 'lambda_init': 1e-08, 'convergence_thresh': 1e-06, 'dist_thresh': 0.1, 'radius': 3, 'dilation_max': 5}, 'tracking': {'min_match_frac': 0.05, 'max_iters': 50, 'C_conf': 0.0, 'Q_conf': 1.5, 'rel_error': 0.001, 'delta_norm': 0.001, 'huber': 1.345, 'match_frac_thresh': 0.333, 'sigma_ray': 0.003, 'sigma_dist': 10.0, 'sigma_pixel': 1.0, 'sigma_depth': 10.0, 'sigma_point': 0.05, 'pixel_border': -10, 'depth_eps': 1e-06, 'filtering_mode': 'weighted_pointmap', 'filtering_score': 'median'}, 'local_opt': {'pin': 1, 'window_size': 1000000.0, 'C_conf': 0.0, 'Q_conf': 1.5, 'min_match_frac': 0.1, 'pixel_border': -10, 'depth_eps': 1e-06, 'max_iters': 10, 'sigma_ray': 0.003, 'sigma_dist': 10.0, 'sigma_pixel': 1.0, 'sigma_depth': 10.0, 'sigma_point': 0.05, 'delta_norm': 1e-08, 'use_cuda': True}, 'retrieval': {'k': 3, 'min_thresh': 0.005}, 'reloc': {'min_match_frac': 0.3, 'strict': True}}
... loading model from checkpoints/MASt3R_ViTLarge_BaseDecoder_512_catmlpdpt_metric.pth
/mnt/c/python_project/WSL/MASt3R-SLAM/thirdparty/mast3r/mast3r/model.py:24: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  ckpt = torch.load(model_path, map_location='cpu')
config/base.yaml
instantiating : AsymmetricMASt3R(enc_depth=24, dec_depth=12, enc_embed_dim=1024, dec_embed_dim=768, enc_num_heads=16, dec_num_heads=12, pos_embed='RoPE100',img_size=(512, 512), head_type='catmlp+dpt', output_mode='pts3d+desc24', depth_mode=('exp', -inf, inf), conf_mode=('exp', 1, inf), patch_embed_cls='PatchEmbedDust3R', two_confs=True, desc_conf_mode=('exp', 0, inf), landscape_only=False)
<All keys matched successfully>
config/base.yaml
Loading retrieval model from checkpoints/MASt3R_ViTLarge_BaseDecoder_512_catmlpdpt_metric_retrieval_trainingfree.pth
/mnt/c/python_project/WSL/MASt3R-SLAM/thirdparty/mast3r/mast3r/retrieval/processor.py:70: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  ckpt = torch.load(modelname, 'cpu')  # TODO from pretrained to download it automatically
/mnt/c/python_project/WSL/MASt3R-SLAM/mast3r_slam/frame.py:116: UserWarning: The given NumPy array is not writable, and PyTorch does not support non-writable tensors. This means writing to this tensor will result in undefined behavior. You may want to copy the array to protect its data or make it writable before converting it to a tensor. This type of warning will be suppressed for the rest of this program. (Triggered internally at ../torch/csrc/utils/tensor_numpy.cpp:206.)
  uimg = torch.from_numpy(img["unnormalized_img"]) / 255.0
Skipped frame 1
FPS: 5.339182840901337
FPS: 4.758726024577935
FPS: 4.661925652676935
FPS: 4.592179711062178
FPS: 4.5253676810531065
FPS: 4.402698903977012
FPS: 4.447971541670871
FPS: 4.507531903670078
FPS: 4.55483376954617
.....
done`

Done Processed Imgui Window:
[image]

Log Folder txt:
[image]

PLY file open in blender:
[image]

Looks like the vertex color store correctly, but vertex position all zero.

bobbyohyeah, 2025-02-27:
Yep, facing the same issue here. I'm running on Windows 10, so I can only use the image folder for input. 

>  UserWarning: The given NumPy array is not writable, and PyTorch does not support non-writable tensors. This means writing to this tensor will result in undefined behavior. You may want to copy the array to protect its data or make it writable before converting it to a tensor. This type of warning will be suppressed for the rest of this program. (Triggered internally at C:\cb\pytorch_1000000000000\work\torch\csrc\utils\tensor_numpy.cpp:212.)
  uimg = torch.from_numpy(img["unnormalized_img"]) / 255.0
Process Process-2:
Traceback (most recent call last):
  File "C:\Users\username\miniconda3\envs\mast3r-slam\Lib\multiprocessing\process.py", line 314, in _bootstrap
    self.run()
  File "C:\Users\username\miniconda3\envs\mast3r-slam\Lib\multiprocessing\process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "C:\workspace\MASt3R-SLAM\mast3r_slam\visualization.py", line 438, in run_visualization
    window.render(current_time, delta)
  File "C:\Users\username\miniconda3\envs\mast3r-slam\Lib\site-packages\moderngl_window\context\base\window.py", line 742, in render
    self.render_func(time, frame_time)
  File "C:\workspace\MASt3R-SLAM\mast3r_slam\visualization.py", line 169, in render
    ptex, ctex, itex = self.textures[keyframe.frame_id]
                       ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^
KeyError: 0
Loading retrieval model from checkpoints/MASt3R_ViTLarge_BaseDecoder_512_catmlpdpt_metric_retrieval_trainingfree.pth
RELOCALIZING against kf  1  and  [0]
Failed to relocalize
RELOCALIZING against kf  1  and  [0]
Failed to relocalize
RELOCALIZING against kf  1  and  [0]
Failed to relocalize
RELOCALIZING against kf  1  and  [0]
Failed to relocalize
RELOCALIZING against kf  1  and  [0]
Failed to relocalize

LeiyuanMa, 2025-02-28:
For the same video, I tried to enter image folder (the images in it are sorted after ffmepg frame extraction), which also showed Failed to relocalize， like below：
`python main.py --dataset ./data/fps_2_1836c39a469c64938c0c75da37491877 --config config/base.yaml --no-viz `
[image]

Then I tried to use the origin video, it works fine
` python main.py --dataset ./data/1836c39a469c64938c0c75da37491877.mov --config config/base.yaml --no-viz`

[image]

godisme1220, 2025-02-28:
Thats weird, both method(images folder & video) won't work for me.

rmurai0610, 2025-02-28:
Ok, the issue seems like that on windows WSL, multiprocessing doesn't work
https://discuss.pytorch.org/t/pytorch-multiprocessing-with-cuda-sets-tensors-to-0/179117
And part of MASt3R's weights were set to zero because of this.

We've just pushed a new branch `windows` which avoids this problem by not spawning the backend process (and forcing single process).
Hopefully, this works now!

atinfinity, 2025-03-05:
> We've just pushed a new branch windows which avoids this problem by not spawning the backend process (and forcing single process).

My environment is as follows.

- Windows 11
- WSL 2(Ubuntu 22.04)

I tried to use `windows` branch. As a result, I could avoid this problem.

closed 2025-03-06 by edexheim

upup0124, 2025-04-02:
> 是的，这里面临同样的问题。我在 Windows 10 上运行 4090，我只能使用图像文件夹进行输入。
> 
> > UserWarning：给定的 NumPy 数组不可写，并且 PyTorch 不支持不可写张量。这意味着写入此张量将导致未定义的行为。在将数组转换为张量之前，您可能希望复制数组以保护其数据或使其可写。对于此程序的其余部分，将禁止显示此类警告。（在 C：\cb\pytorch_1000000000000\work\torch\csrc\utils\tensor_numpy.cpp：212 内部触发。 uimg = torch.from_numpy（img[“unnormalized_img”]） / 255.0 进程 2：回溯（最近调用最后一次）：文件“C：\Users\username\miniconda3\envs\mast3r-slam\Lib\multiprocessing\process.py”，第 314 行，_bootstrap self.run（） 文件“C：\Users\username\miniconda3\envs\mast3r-slam\Lib\multiprocessing\process.py”，第 108 行，在运行 self._target（*self._args， **self._kwargs）文件 “C：\workspace\MASt3R-SLAM\mast3r_slam\visualization.py”，第 438 行，run_visualization window.render（current_time， delta） 文件“C：\Users\username\miniconda3\envs\mast3r-slam\Lib\site-packages\moderngl_window\context\base\window.py”，第 742 行，在 render self.render_func（time， frame_time） 文件中 “C：\workspace\MASt3R-SLAM\mast3r_slam\visualization.py”，第 169 行，在 render ptex， ctex， itex = self.textures[keyframe.frame_id] ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^ KeyError： 0 Loading retrieval model from checkpoints/MASt3R_ViTLarge_BaseDecoder_512_catmlpdpt_metric_retrieval_trainingfree.pth RELOCALIZING against kf 1 and [0] 无法重新本地化 RELOCALIZING against kf 1 and [0] Failed to relocalize against kf 1 and [0] Failed to relocalize against kf 1 and [0] Failed to relocalize针对 kf 1 和 [0] 重新定位失败 根据 kf 1 和 [0] 重新定位失败

我跟您得问题一样，请问您解决了吗

vigneshrk29, 2025-10-06:
Hello, 
the imgui window is not opening when I execute main.py. I am working on a docker container on wsl2 and I have followed all the steps listed:
1) checkout windows
2) Make the following changes listed in this issue: https://github.com/rmurai0610/MASt3R-SLAM/pull/86/files

The imgui folder is not opening despite all the installation steps proceeding smoothly. I am not sure how to proceed? Additionally, I have also tried with images and videos. The raw_images.txt is 0.0 0.0 0.0 0.0 0.0 0.0 0.0 1.0