rmurai0610/MASt3R-SLAM #22: Some kernel is NOT build for Ada Lovelace.

atinfinity, 2025-02-27:
In [README](https://github.com/rmurai0610/MASt3R-SLAM/blob/3619b74617933a9f9b4ea40caf919eae3e7300b9/README.md), there is the following description.

> We run all our experiments on an RTX 4090

But, some CUDA kernel is NOT build for Ada Lovelace.

- [setup.py#L29-L37](https://github.com/rmurai0610/MASt3R-SLAM/blob/3619b74617933a9f9b4ea40caf919eae3e7300b9/setup.py#L29-L37)
- [thirdparty/mast3r/dust3r/croco/models/curope/setup.py#L11-L16](https://github.com/rmurai0610/MASt3R-SLAM/blob/3619b74617933a9f9b4ea40caf919eae3e7300b9/thirdparty/mast3r/dust3r/croco/models/curope/setup.py#L11-L16)

So, user should add `-gencode=arch=compute_89,code=sm_89`(Reference: <https://developer.nvidia.com/cuda-gpus>).

rmurai0610, 2025-03-04:
Hi, thank you for catching this.
We've added `-gencode=arch=compute_89,code=sm_89` but didn't see any performance difference.
Did you notice some in your setup by any chance?

atinfinity, 2025-03-05:
Thank you for your comment. Now, I didn't check performance difference. There may be no difference between the codes generated by CC86 and CC89.

closed 2025-03-06 by rmurai0610