rmurai0610/MASt3R-SLAM #23: How do you handle large video files?

RashikShahjahan, 2025-02-27:
I have noticed that I get  issues related to memory usage, loop detection and max number of frames reached when I tried this on videos longer than 10 minutes (down sampled to 5FPS). I found that the key frames are all stored in memory and there's fixed buffer size. I have played with increasing the buffer size but that affects memory usage. How do you make it scale to videos up to one hour long?

rmurai0610, 2025-02-28:
Hi, it's quite tricky in practice to get dense SLAM working for such a long video (1h) and our current code will require massive changes (like adding sliding window or sub mapping) to support it.

We are not planning to support such features, but if anyone implements it, please let us know!

closed 2025-02-28 by rmurai0610

RashikShahjahan, 2025-02-28:
@rmurai0610 How do you handle real-time streams then?

rmurai0610, 2025-02-28:
Our implementation fixes the upper limit of the keyframes.
You can allocate this dynamically, but you'll face the same OOM issue as the number of keyframe grows.

RomanIvn, 2025-03-02:
> [@rmurai0610](https://github.com/rmurai0610) How do you handle real-time streams then?

For realtime I think that using the WebCamera dataset source is quite simple to modify to accept rtsp or any other method that brings the new frame in RGB format