rmurai0610/MASt3R-SLAM #24: Using known camera intrinsics/extrinsics

vliu15, 2025-02-27:
Hi great work! Could this method be adapted to accept known camera intrinsics/extrinsics (i.e. calibrated, from IMU/SLAM) to output accurate dense depth maps?

LeiyuanMa, 2025-02-28:
https://github.com/rmurai0610/MASt3R-SLAM/issues/14

is this what you want?

rmurai0610, 2025-02-28:
Hi, as @LeiyuanMa mentioned we now support known intrinsics.

We don't have any plan for setting known extrinsics. But you can try and set the known extrinsics if you want by initialising the camera poses with it and disabling camera pose optimisation.

closed 2025-02-28 by rmurai0610

LeiyuanMa, 2025-03-27:
> Hi, as [@LeiyuanMa](https://github.com/LeiyuanMa) mentioned we now support known intrinsics.
> 
> We don't have any plan for setting known extrinsics. But you can try and set the known extrinsics if you want by initialising the camera poses with it and disabling camera pose optimisation.

it seems support known intrinsics with distortion:https://github.com/rmurai0610/MASt3R-SLAM/blob/main/config/intrinsics.yaml, but I wonder do you support estimate intrinsics with distortion ? such as fisheye mono sequences?