rmurai0610/MASt3R-SLAM #34: Is there a way to get the depth for every frame ?

ArghyaChatterjee, 2025-03-02:
Hello,

I was thinking if we can also get the depth frame for every rgb image and log / publish that as image frames (say .png format)?

Thanks.

rmurai0610, 2025-03-03:
Hi, the z-coordinate of pointmap corresponds to the depth, so you can do something similar to this line to get the depth map.
https://github.com/rmurai0610/MASt3R-SLAM/blob/main/mast3r_slam/visualization.py#L378

ArghyaChatterjee, 2025-03-03:
Hello,

Thanks for the reply. It really helps. I wanted to know if this is predicting the depth from any type of deep learned depth estimation module like depth pro, depth anything, mega depth etc or your own depth prediction module ?

Another question is that, I see there are total 3 weights file. What are they doing / predicting ? Are they all loaded at the same time while it's predicting the pose and map?

Thanks.

edexheim, 2025-03-03:
Hello,

Given two images, the MASt3R network (first checkpoint listed) predicts pointmaps for those two images in a common coordinate frame.  It is similar to depth estimation but directly outputs 3D points so that's how we index into the depth above.

The other two checkpoints are used for image retrieval to perform relocalization and loop closure.  Please see [MASt3R-SfM](https://arxiv.org/abs/2409.19152) for more details.

ArghyaChatterjee, 2025-03-04:
Thanks for the reply again. I have a better picture now. I still have 2 more questions from this discussion.

1. So, you are saying the other 2 checkpoints are used for relocalization and loop closure. So basically for relocalization, it's using a deep learned matching method instead of a geametric method (for example GFTT) for which it needs that 2nd weight file. For loop closure, it's replacing the traditional visual loop closure candidate finding method (for example DboW2) with a deep learned one for which it needs the 3rd weight file. Then the question arizes that Is it optimizing any graph with any traditional method (for example GTSAM) once it generates loop closure candidates using deep learned method with the 3rd weight file??

2. Going back to the first question, is it generating / can generate a depth map for each rgb frame (camera publishing at 30hz) or is it only able to generate depth maps for the key frames ( say 40 / 50 depth maps for a 40s video with 1200 RGB frames that I tested)? As I understand, it needs 2 views for generating the pointmap/depth, techniqually it should be able to generate n-1 depth maps for n rgb frames.  Am I correct ?

Thanks.

edexheim, 2025-03-06:
Hello,

1.  The two retrieval checkpoints are essentially performing the same role as DBoW2 by constructing visual words and creating a dictionary of words. We use this to find images that are potential loop closure candidates.  We have a custom global optimizer that aligns all poses using constraints between matched points, so poses are updated with loop closure constraints factored in.
2. Each frame has a depth at some point while running the system.  The network takes in two images, but also generates a pointmap (and depth) for each of those two frames.

Best,
Eric

closed 2025-03-06 by rmurai0610