rmurai0610/MASt3R-SLAM #35: Fast low-quality preview.

RomanIvn, 2025-03-03:
Hi,
Thank you for the amazing contribution.
I've ran my own living room on a vm with RTX4090 & MD Ryzen 9 7950X 16-Core.  CUDA 11.8.  (on runpod.io)
I also set the subsample to 2.    
Using the UI with "Show ALL" very quickly makes the the display slow down and the mouse stops working. without shows all everything is smooth and nice. Anyway I didnt get close to the 15FPS, Max was 5.

Is there a config to enable "Show all" for the point cloud or mesh only? without the colored texture which i feel like causing the lower performance.

rmurai0610, 2025-03-03:
Hi, currently there are no option to render point cloud only. 

I am not familiar with runpod.io; but I assume this is a cloud-vm?
I am not sure why the GUI slows down so much there.
Can you try and run tum/rgbd_dataset_freiburg1_room to make sure that this is not  the problem with your dataset?
We get 11-12 FPS with the GUI set to show all for this dataset.

RomanIvn, 2025-03-03:
Yeah Its cloud VM. `tum/rgbd_dataset_freiburg1_room` runs at 10fps by the console output but the GUI is still laggy when trying to move arround. (very laggy).  
I'll setup the env on a physical machine this week and this will probably solve it.
Thanks!

closed 2025-03-03 by RomanIvn