rmurai0610/MASt3R-SLAM #36: How to Apply This Project to a Custom Dataset of Sequential Images

zyc-bit, 2025-03-03:
Thank you for this amazing work! I have a question: Is it possible to use this project with my own dataset? My dataset consists of several consecutive images. What should I do to make it work? In what format should I prepare my dataset?

zyc-bit, 2025-03-04:
Thank you for your reply. I ran the example, but for me, running the program on my RTX 3060 Ti was very demanding. That's why I referred to this （https://github.com/rmurai0610/MASt3R-SLAM/issues/18#issuecomment-2688233503）solution to get it working. I will try using my own dataset with RGB images only and provide feedback promptly.

zyc-bit, 2025-03-10:
Hello, I can now successfully run my own dataset. I am using the command:  
```
python main.py --dataset <my images folder> --config config/calib.yaml --calib config/intrinsics.yaml
```  
In the final results, the number of output camera poses is very limited, which I believe only includes the camera poses (quaternions) of the keyframes. Is it possible for me to obtain the camera poses for all the images? Could you please advise on what I should modify?

Additionally, I noticed that you have used 3D Gaussian Splatting. Is it possible for me to access the results of the 3D Gaussian Splatting process that you used in your pipeline?

iyhabz, 2025-04-14:
Helo, is there an update on this point?
as i am having the same open points as Zhang with running my own dataset and the need for the output camera poses.