rmurai0610/MASt3R-SLAM #37: Problems with `libtorchcodec`

spagnoloG, 2025-03-03:
Hello I have a problem setting up the environment on ubuntu 22.04.
Let me take you through the steps I took

```bash
conda create -n mast3r-slam python=3.11
conda activate mast3r-slam
```

Then the nvcc version is 12.8:
```bash
(mast3r-slam) gasper@gasper-NUC11BTMi7:~/Documents/MASt3R-SLAM$ nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2025 NVIDIA Corporation
Built on Wed_Jan_15_19:20:09_PST_2025
Cuda compilation tools, release 12.8, V12.8.61
Build cuda_12.8.r12.8/compiler.35404655_0
```

Then I followed and installed the latest possible pytroch version:

```bash

rmurai0610, 2025-03-03:
Hi can you try and comment out the torchcodec in dataloader.py?
We only use that for mp4 import and I want to see if the rest of the install worked ok

spagnoloG, 2025-03-03:
yes, thanks for the quick reply. This actually **solves** the issue!

closed 2025-03-03 by spagnoloG