rmurai0610/MASt3R-SLAM #39: Skipping too may frames with CUDA 12.4 on H100

RashikShahjahan, 2025-03-04:
I ran several tests on both a RTX 4090 with CUDA  version 11.8 and an H100 with CUDA version of 12.4.
I have noticed that the H100 consistently has worse accuracy on the same videos. I printed the output trajectories of each and found that there was an issue with the predicted trajectories not changing for long periods. Can the CUDA version or GPU type affect frame skipping? If so how can we mitigate that?

RashikShahjahan, 2025-03-05:
@rmurai0610 We tried on Ubuntu 22 and on 12.1 with similar results. I ran the main file with default options no-viz.

Also more generally, the results were pretty inconsistent across runs.

rmurai0610, 2025-03-05:
Hi, thanks for trying. 
Do you get inconsistency with 4090 or only for H100?

It might be worth trying the windows branch, which disables backend multiprocessing.

Otherwise, maybe adding compute code for H100 (which is 9.0) following this issue #22 might help.

RashikShahjahan, 2025-03-05:
@rmurai0610 We tried changing compute code to 90. The inconsistency is with both gpus. I  didn't try the windows branch yet.

rmurai0610, 2025-03-05:
Can you let us know which dataset you're running on? 
On standard dataset (fr1) we found results to be quite consistent on 4090 even with the multiprocess mode

RashikShahjahan, 2025-03-05:
@rmurai0610 I was using my own videos. The inconsistency is between different runs of the same video.

rmurai0610, 2025-03-06:
Seems similar to issue #45, and maybe the dataset you're trying on is too hard.
To isolate the problem, can you try running on the datasets we recommend?