rmurai0610/MASt3R-SLAM #44: Is there any way to make real-time better?

ppsohandsome, 2025-03-05:
Thank you for your team's work! I love it. (please ignore my poor English)

I'm implementing mast3r-slam with my Drones.(the real-time stream transport to my pc with 4060 gpu using wifi)

However, I'm wondering if there's any way to make mast3r-slam real-time better, and i can accept a decrease in accuracy or performance.

Maybe use a smaller ckpg or reduce the sampling of features, what is the best way ?

YJ-142150, 2025-03-05:
@ppsohandsome How much FPS does it run when using drone?

ppsohandsome, 2025-03-05:
3 to 4 FPS

ppsohandsome, 2025-03-05:
the drone itself doesn't have compute ability, just serve as a camera while transport stream to pc.

rmurai0610, 2025-03-05:
You can try setting
https://github.com/rmurai0610/MASt3R-SLAM/blob/06ffa85c9728338cd26fe93f7b5f9a96e655e427/mast3r_slam/dataloader.py#L25
to 224

closed 2025-03-05 by rmurai0610