rmurai0610/MASt3R-SLAM #45: Failed to relocalize

YJ-142150, 2025-03-05:
Thanks for your great work!
I tried to run MASt3R-SLAM on custom dataset.
But... it says "Failed to relocalize"

Database retrieval 2 :  {0}
Database retrieval 3 :  {0, 1}
Database retrieval 4 :  {1, 2}
Database retrieval 5 :  {1, 3}
Database retrieval 6 :  {1, 4}
Database retrieval 7 :  {1, 5}
Database retrieval 8 :  {5, 6}
Skipped frame 21
RELOCALIZING against kf  9  and  [8, 7, 6]
Failed to relocalize
RELOCALIZING against kf  9  and  [8, 7, 6]
Failed to relocalize
RELOCALIZING against kf  9  and  [8, 7, 6]
Failed to relocalize
RELOCALIZING against kf  9  and  [8, 7, 6]
Failed to relocalize
RELOCALIZING against kf  9  and  [8, 7, 6]
Failed to relocalize
RELOCALIZING against kf  9  and  [8, 5, 6]
Failed to relocalize

Do you know when this happens?

YJ-142150, 2025-03-05:
I tried it on in the wild data with `--no-calib` version.

rmurai0610, 2025-03-05:
Does running on fr1 dataset work though? I want to isolate whether this is installation issue or dataset issue.

It could be that the video you're running the system on is too challenging. We need the scene to be more or less static and have continuous camera trajectory as it's a SLAM system

closed 2025-03-06 by YJ-142150