rmurai0610/MASt3R-SLAM #48: If there is anyway to support png (rgba-4channel) ?

ppsohandsome, 2025-03-06:
These day i have disturbed your team many times, sorry about that...and that is all because the appreciate of your team's work

I want to let this slam system deal with a 4-channel image (including Alpha---somewhere is transparent). While changing the `mast3r_utils.resize_img` function, I finally found out that maybe Torch does not support RGBA.

so I give up changing package mast3r to make it support RGBA.

Instead, I come up with 2 alternative ways.

1.skip some pix colour while matching feature----for example, fill transparent with specific colour like white or black.

2.skip some location while matching feature----for example, set a mask of the image.

please let me know what is the best way, thanks so much.

by the way, I'm doing my Undergraduate thesis bases on drone Obstacle Avoidance and object detection (the drone fly behind me and 3d reconstruction without "me" show on the 3d map)

rmurai0610, 2025-03-06:
Hi, no worries.
I'm assuming that the alpha channel is some sort of segmentation mask of a person?
If so, I'll recommend option 2 of masking the image.

Also when visualising, you can set the confidence of masked out points to 0 so that they are not shown in the map.

closed 2025-03-06 by rmurai0610

laiyuzhi, 2025-05-13:
Hi, I have same question. Similarly I feel that it should be done without considering the correspondence of the masked feature points. But  for example, should I set the D of these points to 0 or something else? Can you tell me in a little more detail how to skip these feature points?