rmurai0610/MASt3R-SLAM #54: Can the system cope with pure rotations

antoan, 2025-03-09:
Thank you for making this public. 

Forgive my ignorance, but I'm curious: do the strong priors enable the system to handle a camera moving solely in place?
I'd appreciate any thoughts or feedback you have on this.

Thanks in advance.

edexheim, 2025-03-11:
Hello,

Yes, the system can handle pure rotation.  The optimization for tracking and mapping both use depth information from MASt3R to handle rotation.  For example, this parameter https://github.com/rmurai0610/MASt3R-SLAM/blob/c3d0d5b67bf51d558d7640ff6032407f68041f92/config/base.yaml#L26 is the standard deviation of error in predicted distance from MASt3R.  If one wants to trust the distance more, this could be decreased for example, but we found the current parameters to work well.

Eric

antoan, 2025-03-11:
Thank you so much!

antoan, 2025-04-03:
@edexheim

I wanted to share some recent results we’ve obtained.

Below are the outcomes from a 0–170° panning scan, captured in 5° increments — without applying any intrinsic parameters.

At the moment, the estimated camera pose appears quite unstable, with displacements in the range of several tens of centimeters, despite the camera only rotating in place. You can see this clearly in the attached plot.

The following command was used to run the setup:

`conda activate mast3r-slam && python3 main.py --dataset ~/dev/cv/MASt3R-SLAM/datasets/flox/png_sorted/ --config config/base.yaml`

We're working with a ceiling-mounted cameras industrial poultry broiler hangars here in the UK.

I plan to try the `sigma_dist: 1e+1` setting that was suggested earlier. In the meantime, do you have any other suggestions or parameters I might experiment with to improve stability?

Thanks again for the great work on the project!

[poses.txt](https://github.com/user-attachments/files/19594564/poses.txt)
[image]
https://github.com/user-attachments/assets/2a344051-7ec7-4f48-8daa-606b8a3e1fc4

antoan, 2025-04-03:
https://github.com/user-attachments/assets/94cb12c5-438e-4f4f-a750-6d18823cec32