rmurai0610/MASt3R-SLAM #58: ConnectionResetError: [Errno 104] Connection reset by peer Error

amitlisha, 2025-03-11:
I'm trying to run inference on a folder of images using -
python main.py --dataset ./images/ --config config/base.yaml --no-viz

and I'm getting these errors - 
  uimg = torch.from_numpy(img["unnormalized_img"]) / 255.0
Skipped frame 1
RELOCALIZING against kf  1  and  [0]
Success! Relocalized
Traceback (most recent call last):
  File "/home/sagemaker-user/MASt3R-SLAM/main.py", line 315, in <module>
    eval.save_reconstruction(
  File "/home/sagemaker-user/MASt3R-SLAM/mast3r_slam/evaluate.py", line 62, in save_reconstruction
    keyframe.get_average_conf().cpu().numpy().astype(np.float32).reshape(-1)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sagemaker-user/MASt3R-SLAM/mast3r_slam/frame.py", line 108, in get_average_conf
    return self.C / self.N if self.C is not None else None
           ~~~~~~~^~~~~~~~
RuntimeError: CUDA error: the provided PTX was compiled with an unsupported toolchain.
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

Process Process-2:
Traceback (most recent call last):
  File "/home/sagemaker-user/.conda/envs/mast3r-slam/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/sagemaker-user/.conda/envs/mast3r-slam/lib/python3.11/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/sagemaker-user/MASt3R-SLAM/main.py", line 83, in run_backend
    mode = states.get_mode()
           ^^^^^^^^^^^^^^^^^
  File "/home/sagemaker-user/MASt3R-SLAM/mast3r_slam/frame.py", line 200, in get_mode
    with self.lock:
  File "/home/sagemaker-user/.conda/envs/mast3r-slam/lib/python3.11/multiprocessing/managers.py", line 1056, in __enter__
    return self._callmethod('acquire')
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sagemaker-user/.conda/envs/mast3r-slam/lib/python3.11/multiprocessing/managers.py", line 822, in _callmethod
    kind, result = conn.recv()
                   ^^^^^^^^^^^
  File "/home/sagemaker-user/.conda/envs/mast3r-slam/lib/python3.11/multiprocessing/connection.py", line 250, in recv
    buf = self._recv_bytes()
          ^^^^^^^^^^^^^^^^^^
  File "/home/sagemaker-user/.conda/envs/mast3r-slam/lib/python3.11/multiprocessing/connection.py", line 430, in _recv_bytes
    buf = self._recv(4)
          ^^^^^^^^^^^^^
  File "/home/sagemaker-user/.conda/envs/mast3r-slam/lib/python3.11/multiprocessing/connection.py", line 395, in _recv
    chunk = read(handle, remaining)
            ^^^^^^^^^^^^^^^^^^^^^^^
ConnectionResetError: [Errno 104] Connection reset by peer

I'm using -
Nvidia A10G
Cuda 12.5

I would appreciate the help, thanks.

rmurai0610, 2025-03-12:
Hi, can you tell me which dataset you are testing on? 
Also, if you can try the windows branch (it disables multi-processing which seems to be causing problem for few people) that would be great.

amitlisha, 2025-03-15:
I got it to work by downgrading CUDA to 11.8

closed 2025-03-15 by rmurai0610