rmurai0610/MASt3R-SLAM #59: System establish

Aoko955, 2025-03-12:
Thank you for your good work! The demo video on your project page showed that your system could capture pictures on the mobile phone and transferred  them to the server to do the reconstruction. Could you please share related code, such as the code for communication between mobile phone and server?

rmurai0610, 2025-03-12:
Hi, we used https://droidcam.app/ 
Once droidcam is setup, you can use the webcam option
`python main.py --dataset webcam`

Aoko955, 2025-03-13:
Thank you for your quick reply! I have set up the droidcam and connected successfully by running "python main.py" --dataset webcam. How can I stop the reconstruction progress and save the point cloud document?

Aoko955, 2025-03-14:
@rmurai0610

closed 2025-03-16 by Aoko955