rmurai0610/MASt3R-SLAM #60: pip install -e thirdparty/mast3r error. Cannot install "curope"

DtiSignal, 2025-03-12:
Hello everyone. Help solve the error. Cannot install "curope"

CUDA 12.4. GPU 4090. python 3.11. Thanks for any tips

[image]

sumitsarkar1, 2025-03-13:
1. do `conda install nvidia/label/cuda-12.4.0::cuda-toolkit` to your conda env
2. check` which nvcc `it should show `anaconda3/envs/mast3r-slam/bin/nvcc`

rmurai0610, 2025-03-14:
Hi, can you try using WSL on the windows branch? Few people now have managed to get it running on Windows using it.

sumitsarkar1, 2025-03-16:
@DtiSignal extremely sorry..I completely ignored the fact u r on a Windows machine...`which nvcc` wont work for a Windows machine..I did the installation on a Ubuntu machine

sumitsarkar1, 2025-03-16:
@DtiSignal https://www.youtube.com/watch?v=TK8DK19o6YQ

DtiSignal, 2025-03-18:
Thank you very much everyone for your help! I met "WSL" for the first time. Everything worked perfectly. The video tutorial from [sumitsarkar1](https://github.com/rmurai0610/MASt3R-SLAM/issues/60#issuecomment-2727377674) sped up the process. Thanks to the developers for a great project!

closed 2025-03-18 by DtiSignal