rmurai0610/MASt3R-SLAM #62: CUDA Runtime Error

Schwartz-Zha, 2025-03-20:
I'm trying to tun the demo but constantly got this error.  My command is python main.py --dataset datasets/tum/rgbd_dataset_freiburg1_room/ --config config/calib.yaml --no-viz, and the log info is listed below.
`RELOCALIZING against kf  49  and  [48, 16, 47]
Success! Relocalized
FPS: 13.627139871677088
Cholesky failed 631
RELOCALIZING against kf  50  and  [49, 48, 46]
Success! Relocalized
Cholesky failed 637
RELOCALIZING against kf  51  and  [50, 20, 21]
Success! Relocalized
Cholesky failed 643
RELOCALIZING against kf  52  and  [51, 20, 50]
Success! Relocalized
Cholesky failed 649
RELOCALIZING against kf  53  and  [50, 15, 52]
Success! Relocalized
Cholesky failed 655
RELOCALIZING against kf  54  and  [53, 50, 49]
Success! Relocalized
FPS: 13.551781993027333
Cholesky failed 661
RELOCALIZING against kf  55  and  [54, 53, 47]
Success! Relocalized
Cholesky failed 667
RELOCALIZING against kf  56  and  [55, 54, 46]
Success! Relocalized
Cholesky failed 673
RELOCALIZING against kf  57  and  [56, 55, 47]
Success! Relocalized
Cholesky failed 679
RELOCALIZING against kf  58  and  [57, 56, 55]
Success! Relocalized
Traceback (most recent call last):
  File "/data/MASt3R-SLAM/main.py", line 315, in <module>
    eval.save_reconstruction(
  File "/data/MASt3R-SLAM/mast3r_slam/evaluate.py", line 62, in save_reconstruction
    keyframe.get_average_conf().cpu().numpy().astype(np.float32).reshape(-1)
  File "/data/MASt3R-SLAM/mast3r_slam/frame.py", line 108, in get_average_conf
    return self.C / self.N if self.C is not None else None
RuntimeError: CUDA error: the provided PTX was compiled with an unsupported toolchain.
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

Process Process-2:
Traceback (most recent call last):
  File "/usr/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/usr/lib/python3.10/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/data/MASt3R-SLAM/main.py", line 83, in run_backend
    mode = states.get_mode()
  File "/data/MASt3R-SLAM/mast3r_slam/frame.py", line 200, in get_mode
    with self.lock:
  File "/usr/lib/python3.10/multiprocessing/managers.py", line 1052, in __enter__
    return self._callmethod('acquire')
  File "/usr/lib/python3.10/multiprocessing/managers.py", line 833, in _callmethod
    raise convert_to_error(kind, result)
multiprocessing.managers.RemoteError: 
---------------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.10/multiprocessing/managers.py", line 260, in serve_client
    self.id_to_local_proxy_obj[ident]
KeyError: '7fe9359a2840'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.10/multiprocessing/managers.py", line 262, in serve_client
    raise ke
  File "/usr/lib/python3.10/multiprocessing/managers.py", line 256, in serve_client
    obj, exposed, gettypeid = id_to_obj[ident]
KeyError: '7fe9359a2840'
---------------------------------------------------------------------------`

Schwartz-Zha, 2025-03-20:
I'm on Ubuntu 22.04 and CUDA 12.1, actually I closely followed the installation method in README.md and met no issues during installation. I cannot figure out how this error emerged, appreciations to all kinds of help!

```(venv_MS) zhajiajun@video-sv4xf-1451203-worker-0:/data/MASt3R-SLAM$ python -c "import torch; print(torch.__version__)"
2.5.1+cu121
(venv_MS) zhajiajun@video-sv4xf-1451203-worker-0:/data/MASt3R-SLAM$ nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Tue_Feb__7_19:32:13_PST_2023
Cuda compilation tools, release 12.1, V12.1.66
Build cuda_12.1.r12.1/compiler.32415258_0
(venv_MS) zhajiajun@video-sv4xf-1451203-worker-0:/data/MASt3R-SLAM$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04 LTS
Release:        22.04
Codename:       jammy
```

Schwartz-Zha, 2025-03-23:
After some debugging, I found that this error only occurs when

closed 2025-03-23 by Schwartz-Zha

manhha1402, 2025-04-10:
@Schwartz-Zha I got the same issue. Can you please help me what is solution for it?

fumseckk, 2025-06-10:
I am having the same issue. Can you please share your solution ?