rmurai0610/MASt3R-SLAM #68: Error when compiling on Windows

Hillowwold, 2025-03-28:
Hi,
I got error when I was trying to compiling the code on Windows (Not WSL), and I got some error like.

```

Building wheels for collected packages: MAST3R-SLAM, lietorch
  Building editable for MAST3R-SLAM (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building editable for MAST3R-SLAM (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [160 lines of output]
      running editable_wheel
      creating C:\Users\ldl-f\AppData\Local\Temp\pip-wheel-itmwcd81\.tmp-ps9x_d2l\MAST3R_SLAM.egg-info
      writing C:\Users\ldl-f\AppData\Local\Temp\pip-wheel-itmwcd81\.tmp-ps9x_d2l\MAST3R_SLAM.egg-info\PKG-INFO
      writing dependency_links to C:\Users\ldl-f\AppData\Local\Temp\pip-wheel-itmwcd81\.tmp-ps9x_d2l\MAST3R_SLAM.egg-info\dependency_links.txt
      writing requirements to C:\Users\ldl-f\AppData\Local\Temp\pip-wheel-itmwcd81\.tmp-ps9x_d2l\MAST3R_SLAM.egg-info\requires.txt
      writing top-level names to C:\Users\ldl-f\AppData\Local\Temp\pip-wheel-itmwcd81\.tmp-ps9x_d2l\MAST3R_SLAM.egg-info\top_level.txt
      writing manifest file 'C:\Users\ldl-f\AppData\Local\Temp\pip-wheel-itmwcd81\.tmp-ps9x_d2l\MAST3R_SLAM.egg-info\SOURCES.txt'
      C:\Users\ldl-f\anaconda3\envs\mast3r-slam\Lib\site-packages\torch\utils\cpp_extension.py:497: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
        warnings.warn(msg.format('we could not find ninja.'))
      reading manifest file 'C:\Users\ldl-f\AppData\Local\Temp\pip-wheel-itmwcd81\.tmp-ps9x_d2l\MAST3R_SLAM.egg-info\SOURCES.txt'
      adding license file 'LICENSE.md'
      writing manifest file 'C:\Users\ldl-f\AppData\Local\Temp\pip-wheel-itmwcd81\.tmp-ps9x_d2l\MAST3R_SLAM.egg-info\SOURCES.txt'
      creating 'C:\Users\ldl-f\AppData\Local\Temp\pip-wheel-itmwcd81\.tmp-ps9x_d2l\MAST3R_SLAM-0.0.1.dist-info'
      C:\Users\ldl-f\anaconda3\envs\mast3r-slam\Lib\site-packages\setuptools\command\bdist_wheel.py:108: RuntimeWarning: Config variable 'Py_DEBUG' is unset, Python ABI tag may be incorrect
        if get_flag("Py_DEBUG", hasattr(sys, "gettotalrefcount"), warn=(impl == "cp")):
      creating C:\Users\ldl-f\AppData\Local\Temp\pip-wheel-itmwcd81\.tmp-ps9x_d2l\MAST3R_SLAM-0.0.1.dist-info\WHEEL
      running build_py
      running build_ext
      C:\Users\ldl-f\anaconda3\envs\mast3r-slam\Lib\site-packages\torch\utils\cpp_extension.py:382: UserWarning: Error checking compiler version for cl: [WinError 2] The system cannot find the file specified
        warnings.warn(f'Error checking compiler version for {compiler}: {error}')
      building 'mast3r_slam_backends' extension
      creating C:\Users\ldl-f\AppData\Local\Temp\tmpl8sqcrcp.build-temp\Release\mast3r_slam\backend\src
      "C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.43.34808\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -IC:\Users\ldl-f\Desktop\Master_Theis_Jiuming_Yanling\MASt3R-SLAM\mast3r_slam/backend/include -IC:\Users\ldl-f\Desktop\Master_Theis_Jiuming_Yanling\MASt3R-SLAM\thirdparty/eigen -IC:\Users\ldl-f\anaconda3\envs\mast3r-slam\Lib\site-packages\torch\include -IC:\Users\ldl-f\anaconda3\envs\mast3r-slam\Lib\site-packages\torch\include\torch\csrc\api\include -IC:\Users\ldl-f\anaconda3\envs\mast3r-slam\Lib\site-packages\torch\include\TH -IC:\Users\ldl-f\anaconda3\envs\mast3r-slam\Lib\site-packages\torch\include\THC "-IC:\Program Files\NVIDIA GPU Computing Toolkit\CUDA3\include" -IC:\Users\ldl-f\anaconda3\envs\mast3r-slam\include -IC:\Users\ldl-f\anaconda3\envs\mast3r-slam\Include "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.43.34808\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Auxiliary\VS\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\um" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\shared" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\winrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\cppwinrt" /EHsc /Tpmast3r_slam/backend/src/gn.cpp /FoC:\Users\ldl-f\AppData\Local\Temp\tmpl8sqcrcp.build-temp\Release\mast3r_slam\backend\src\gn.obj /MD /wd4819 /wd4251 /wd4244 /wd4267 /wd4275 /wd4018 /wd4190 /wd4624 /wd4067 /wd4068 /EHsc -O3 -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=mast3r_slam_backends -D_GLIBCXX_USE_CXX11_ABI=0 /std:c++17
      cl : Command line warning D9002 : ignoring unknown option '-O3'
      gn.cpp
      "C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA3\bin\nvcc" -c mast3r_slam/backend/src/gn_kernels.cu -o C:\Users\ldl-f\AppData\Local\Temp\tmpl8sqcrcp.build-temp\Release\mast3r_slam\backend\src\gn_kernels.obj -IC:\Users\ldl-f\Desktop\Master_Theis_Jiuming_Yanling\MASt3R-SLAM\mast3r_slam/backend/include -IC:\Users\ldl-f\Desktop\Master_Theis_Jiuming_Yanling\MASt3R-SLAM\thirdparty/eigen -IC:\Users\ldl-f\anaconda3\envs\mast3r-slam\Lib\site-packages\torch\include -IC:\Users\ldl-f\anaconda3\envs\mast3r-slam\Lib\site-packages\torch\include\torch\csrc\api\include -IC:\Users\ldl-f\anaconda3\envs\mast3r-slam\Lib\site-packages\torch\include\TH -IC:\Users\ldl-f\anaconda3\envs\mast3r-slam\Lib\site-packages\torch\include\THC "-IC:\Program Files\NVIDIA GPU Computing Toolkit\CUDA3\include" -IC:\Users\ldl-f\anaconda3\envs\mast3r-slam\include -IC:\Users\ldl-f\anaconda3\envs\mast3r-slam\Include "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.43.34808\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Auxiliary\VS\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\um" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\shared" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\winrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\cppwinrt" -Xcudafe --diag_suppress=dll_interface_conflict_dllexport_assumed -Xcudafe --diag_suppress=dll_interface_conflict_none_assumed -Xcudafe --diag_suppress=field_without_dll_interface -Xcudafe --diag_suppress=base_class_has_different_dll_interface -Xcompiler /EHsc -Xcompiler /wd4068 -Xcompiler /wd4067 -Xcompiler /wd4624 -Xcompiler /wd4190 -Xcompiler /wd4018 -Xcompiler /wd4275 -Xcompiler /wd4267 -Xcompiler /wd4244 -Xcompiler /wd4251 -Xcompiler /wd4819 -Xcompiler /MD -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr -O3 -gencode=arch=compute_60,code=sm_60 -gencode=arch=compute_61,code=sm_61 -gencode=arch=compute_70,code=sm_70 -gencode=arch=compute_75,code=sm_75 -gencode=arch=compute_80,code=sm_80 -gencode=arch=compute_86,code=sm_86 -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=mast3r_slam_backends -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++17 --use-local-env
      C:\Users\ldl-f\Desktop\Master_Theis_Jiuming_Yanling\MASt3R-SLAM\mast3r_slam\backend\src\gn_kernels.cu(344): warning #177-D: label "https" was declared but never referenced
          https:
          ^

      Remark: The warnings can be suppressed with "-diag-suppress <warning-number>"

      C:\Users\ldl-f\Desktop\Master_Theis_Jiuming_Yanling\MASt3R-SLAM\mast3r_slam\backend\src\gn_kernels.cu(344): warning #177-D: label "https" was declared but never referenced
          https:
          ^

      Remark: The warnings can be suppressed with "-diag-suppress <warning-number>"

      C:\Users\ldl-f\Desktop\Master_Theis_Jiuming_Yanling\MASt3R-SLAM\mast3r_slam\backend\src\gn_kernels.cu(344): warning #177-D: label "https" was declared but never referenced
          https:
          ^

      Remark: The warnings can be suppressed with "-diag-suppress <warning-number>"

      C:\Users\ldl-f\Desktop\Master_Theis_Jiuming_Yanling\MASt3R-SLAM\mast3r_slam\backend\src\gn_kernels.cu(344): warning #177-D: label "https" was declared but never referenced
          https:
          ^

      Remark: The warnings can be suppressed with "-diag-suppress <warning-number>"

      C:\Users\ldl-f\Desktop\Master_Theis_Jiuming_Yanling\MASt3R-SLAM\mast3r_slam\backend\src\gn_kernels.cu(344): warning #177-D: label "https" was declared but never referenced
          https:
          ^

      Remark: The warnings can be suppressed with "-diag-suppress <warning-number>"

[... cut ...]
  Created wheel for lietorch: filename=lietorch-0.2-cp311-cp311-win_amd64.whl size=2811839 sha256=2010203b875689f68604a13d7bfbdb21d5796d7212917816db8cfd82089a9e9d
  Stored in directory: C:\Users\ldl-f\AppData\Local\Temp\pip-ephem-wheel-cache-pmfrdwfg\wheels\ab\b0\8a\462452dd05f5bd59fa6a0359ac8b79099168de1c10ac279b52
Successfully built lietorch
Failed to build MAST3R-SLAM
ERROR: Failed to build installable wheels for some pyproject.toml based projects (MAST3R-SLAM)

```
I am using 4090 with CUDA toolkit 12.4. Out of Some reasons, I can not use WSL, so do you guys have any idea about how to solve it? Thank you!

rmurai0610, 2025-03-28:
Hi, sorry we only developed the system on Ubuntu and the code probably would not work on Windows directly unless WLS is used.

closed 2025-03-28 by rmurai0610

Hillowwold, 2025-04-01:
> Hi, I was able to fork the repository and get it running on windows 11, pytorch 2.6 and cuda 12.4

Hi, have you done any changes with the code?

Dhuldhoyavarun, 2025-04-01:
Yes, you can clone the main branch of https://github.com/Dhuldhoyavarun/MASt3R-SLAM and follow the readme there.

Enzo-Kerkhof, 2025-06-24:
@Dhuldhoyavarun Very nice, seems like the code is compiling for me. However when I try tro run the code I have one finale issue:
```cmd
...\.conda\envs\mast3r-slam\Lib\site-packages\glfw\__init__.py:917: GLFWError: (65542) b'EGL: Library not found'
  warnings.warn(message, GLFWError)
Failed to create window
```
How did you overcome this? I suppose I can run the code headless but I need the visualization.