rmurai0610/MASt3R-SLAM #69: libeigen  cloning failed

Afloat16, 2025-03-28:
Since I am a rented server and cannot use a VPN to connect to the Internet (the server can access github but not gitlab), I would like to ask about the method and version of libeigen manual installation, I will use my local download package and upload it to the server for manual installation. Here's the error that comes up when you run “pip install --no-build-isolation -e.”:

(mast3r-slam) root@autodl-container-89914fa1bb-8fe96ba8:~/MASt3R-SLAM# pip install --no-build-isolation -e .
Looking in indexes: http://mirrors.aliyun.com/pypi/simple
Obtaining file:///root/MASt3R-SLAM
  Checking if build backend supports build_editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Collecting lietorch@ git+https://github.com/princeton-vl/lietorch.git (from MAST3R-SLAM==0.0.1)
  Cloning https://github.com/princeton-vl/lietorch.git to /tmp/pip-install-5k94lux9/lietorch_584c3c8aaccc484b9d82f6aa943a1761
  Running command git clone --filter=blob:none --quiet https://github.com/princeton-vl/lietorch.git /tmp/pip-install-5k94lux9/lietorch_584c3c8aaccc484b9d82f6aa943a1761
  Resolved https://github.com/princeton-vl/lietorch.git to commit 0fa9ce8ffca86d985eca9e189a99690d6f3d4df6
  Running command git submodule update --init --recursive -q
  fatal: unable to access 'https://gitlab.com/libeigen/eigen.git/': Failed to connect to gitlab.com port 443 after 131199 ms: Connection timed out
  fatal: clone of 'https://gitlab.com/libeigen/eigen.git' into submodule path '/tmp/pip-install-5k94lux9/lietorch_584c3c8aaccc484b9d82f6aa943a1761/eigen' failed
  Failed to clone 'eigen'. Retry scheduled
  fatal: unable to access 'https://gitlab.com/libeigen/eigen.git/': Failed to connect to gitlab.com port 443 after 131052 ms: Connection timed out
  fatal: clone of 'https://gitlab.com/libeigen/eigen.git' into submodule path '/tmp/pip-install-5k94lux9/lietorch_584c3c8aaccc484b9d82f6aa943a1761/eigen' failed
  Failed to clone 'eigen' a second time, aborting
  error: subprocess-exited-with-error
  
  × git submodule update --init --recursive -q did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

Looking forward to your answer, thank you！

rmurai0610, 2025-03-28:
Hi, sorry we don't have any plan on supporting such feature (as Eigen is installed both by lietorch and by us).
You can try and locally clone/download Eigen into the appropiate directory for both our code and lietorch (`thirdparty/eigen` for MASt3R-SLAM), remove the eigen dependency from the submodule and build the system.

closed 2025-03-28 by rmurai0610

Afloat16, 2025-03-30:
Okay, thank you for your answer.