rmurai0610/MASt3R-SLAM #73: I want run mast3r-slam with my RTX 5070 ti GPU.

PR5GRAMM2R, 2025-04-06:
I tried mast3r-slam with my new GPU RTX 5070 ti.

I installed Ubuntu 24.04 and Custom Linux Kernel 6.14.0 using mainline.

I needed Linux Kernel 6.14.0 to use RTX 5070 ti and installed newest Nvidia Driver.

And I installed CUDA 12.4 and following pytorches that specified in your git repository.

This following codes was worked successfully in anaconda's virtual environment.

pip install -e thirdparty/mast3r
pip install -e thirdparty/in3d
pip install --no-build-isolation -e .
pip install torchcodec==0.1

I got tum Dataset and runned this following executing commend.

python main.py --dataset datasets/tum/rgbd_dataset_freiburg1_room/ --config config/calib.yaml

But the error occured which can't find the device (RTX 5070 ti).

I asked to ChatGPT andit answered that Pytorch 2.5.1 does not support sm_120.

So, I compiled Pytorch 2.5.1 with fallbacked with sm_90 and installed the new Pytorch and comparible torchvision and torchaudio.

And with the support of ChatGPT, I edited 
MASt3R-SLAM/mast3r_slam/backend/src/gn_kernels.cu file .

I changed each line of

delta_norm = torch::linalg::linalg_norm(dx, std::optional<c10::Scalar>(), {}, false, {});

to this.

delta_norm = torch::linalg::linalg_norm(
    dx,
    c10::nullopt,                      // no ord
    at::OptionalIntArrayRef(),        // dims
    false,                            // keepdim
    c10::optional<c10::ScalarType>()  // dtype
);

And I cloned lietorch myself and installed it.

And I runned mast3r-slam but doesn't worked.

Is there other method to run mast3r-slam with RTX 50 esries?

I am suffering with running mast3r-slam with rtx 5070 ti. I spent working with this for 8 days. It would be greatful to solve this problem. Thanks for reading my issue.

PR5GRAMM2R, 2025-04-30:
Thank you for your suggestions.

I first installed cuda 12.8 and pytorch 2.8.0 .
Second, I modified the changes the you suggested in 
https://github.com/rmurai0610/MASt3R-SLAM/pull/86/commits/a9ab9b6394f39599fc47426d0c88096b9d4c6193
https://github.com/rmurai0610/MASt3R-SLAM/pull/86/commits/735a8767ce4bc8ba7b0d3620b27380c5e25a652e
 and saved the modified files.

And Then I moved to Mast3r-slam Directory and did these things.

pip install -e thirdparty/mast3r

pip install -e thirdparty/in3d

It worked successfully.

But, in the installation of pip install --no-build-isolation -e . 
, the errors occured with lietorch.

This is the log of installation on CLI of Ubuntu 22.04 of WSL2.

Thanks for your answer. I'll be really appreciate for your support.

=========================================================================

(mast3r502) juhwan@DESKTOPJUHWAN:~/MASt3R-SLAM$ pip install --no-build-isolation -e .
Obtaining file:///home/juhwan/MASt3R-SLAM
  Checking if build backend supports build_editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Collecting lietorch@ git+https://github.com/princeton-vl/lietorch.git (from MAST3R-SLAM==0.0.1)
  Cloning https://github.com/princeton-vl/lietorch.git to /tmp/pip-install-r_4u6dqu/lietorch_5f8385db916f4dac97d08fce28c8de95
  Running command git clone --filter=blob:none --quiet https://github.com/princeton-vl/lietorch.git /tmp/pip-install-r_4u6dqu/lietorch_5f8385db916f4dac97d08fce28c8de95
  Resolved https://github.com/princeton-vl/lietorch.git to commit 0fa9ce8ffca86d985eca9e189a99690d6f3d4df6
  Running command git submodule update --init --recursive -q
  Preparing metadata (setup.py) ... done
Requirement already satisfied: numpy==1.26.4 in /home/juhwan/anaconda3/envs/mast3r502/lib/python3.11/site-packages (from MAST3R-SLAM==0.0.1) (1.26.4)
Requirement already satisfied: einops in /home/juhwan/anaconda3/envs/mast3r502/lib/python3.11/site-packages (from MAST3R-SLAM==0.0.1) (0.8.1)
Collecting pyrealsense2 (from MAST3R-SLAM==0.0.1)
  Using cached pyrealsense2-2.55.1.6486-cp311-cp311-manylinux1_x86_64.whl.metadata (1.8 kB)
Collecting evo (from MAST3R-SLAM==0.0.1)
  Using cached evo-1.31.1-py3-none-any.whl.metadata (13 kB)
Collecting natsort (from MAST3R-SLAM==0.0.1)
  Using cached natsort-8.4.0-py3-none-any.whl.metadata (21 kB)
Collecting plyfile (from MAST3R-SLAM==0.0.1)
  Using cached plyfile-1.1-py3-none-any.whl.metadata (2.1 kB)
Collecting argcomplete (from evo->MAST3R-SLAM==0.0.1)
  Using cached argcomplete-3.6.2-py3-none-any.whl.metadata (16 kB)
Collecting colorama>=0.3 (from evo->MAST3R-SLAM==0.0.1)
  Using cached colorama-0.4.6-py2.py3-none-any.whl.metadata (17 kB)
Requirement already satisfied: matplotlib>=3.6.0 in /home/juhwan/anaconda3/envs/mast3r502/lib/python3.11/site-packages (from evo->MAST3R-SLAM==0.0.1) (3.10.1)
Collecting numexpr>=2.7.3 (from evo->MAST3R-SLAM==0.0.1)
  Using cached numexpr-2.10.2-cp311-cp311-manylinux_2_27_x86_64.manylinux_2_28_x86_64.whl.metadata (8.1 kB)
Requirement already satisfied: pandas in /home/juhwan/anaconda3/envs/mast3r502/lib/python3.11/site-packages (from evo->MAST3R-SLAM==0.0.1) (2.2.3)
Requirement already satisfied: pillow in /home/juhwan/anaconda3/envs/mast3r502/lib/python3.11/site-packages (from evo->MAST3R-SLAM==0.0.1) (11.0.0)
Requirement already satisfied: pygments in /home/juhwan/anaconda3/envs/mast3r502/lib/python3.11/site-packages (from evo->MAST3R-SLAM==0.0.1) (2.19.1)
Requirement already satisfied: pyyaml in /home/juhwan/anaconda3/envs/mast3r502/lib/python3.11/site-packages (from evo->MAST3R-SLAM==0.0.1) (6.0.2)
Collecting rosbags>=0.9.20 (from evo->MAST3R-SLAM==0.0.1)
  Using cached rosbags-0.10.9-py3-none-any.whl.metadata (5.0 kB)
Requirement already satisfied: scipy>=1.2 in /home/juhwan/anaconda3/envs/mast3r502/lib/python3.11/site-packages (from evo->MAST3R-SLAM==0.0.1) (1.15.2)
Collecting seaborn>=0.9 (from evo->MAST3R-SLAM==0.0.1)
  Using cached seaborn-0.13.2-py3-none-any.whl.metadata (5.4 kB)
Requirement already satisfied: contourpy>=1.0.1 in /home/juhwan/anaconda3/envs/mast3r502/lib/python3.11/site-packages (from matplotlib>=3.6.0->evo->MAST3R-SLAM==0.0.1) (1.3.2)
Requirement already satisfied: cycler>=0.10 in /home/juhwan/anaconda3/envs/mast3r502/lib/python3.11/site-packages (from matplotlib>=3.6.0->evo->MAST3R-SLAM==0.0.1) (0.12.1)
Requirement already satisfied: fonttools>=4.22.0 in /home/juhwan/anaconda3/envs/mast3r502/lib/python3.11/site-packages (from matplotlib>=3.6.0->evo->MAST3R-SLAM==0.0.1) (4.57.0)
Requirement already satisfied: kiwisolver>=1.3.1 in /home/juhwan/anaconda3/envs/mast3r502/lib/python3.11/site-packages (from matplotlib>=3.6.0->evo->MAST3R-SLAM==0.0.1) (1.4.8)
Requirement already satisfied: packaging>=20.0 in /home/juhwan/anaconda3/envs/mast3r502/lib/python3.11/site-packages (from matplotlib>=3.6.0->evo->MAST3R-SLAM==0.0.1) (25.0)
Requirement already satisfied: pyparsing>=2.3.1 in /home/juhwan/anaconda3/envs/mast3r502/lib/python3.11/site-packages (from matplotlib>=3.6.0->evo->MAST3R-SLAM==0.0.1) (3.2.3)
Requirement already satisfied: python-dateutil>=2.7 in /home/juhwan/anaconda3/envs/mast3r502/lib/python3.11/site-packages (from matplotlib>=3.6.0->evo->MAST3R-SLAM==0.0.1) (2.9.0.post0)
Collecting lz4 (from rosbags>=0.9.20->evo->MAST3R-SLAM==0.0.1)
  Using cached lz4-4.4.4-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (3.8 kB)
[... 1032 lines cut ...]
[... cut ...]
      error: command '/usr/bin/g++' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for lietorch
  Running setup.py clean for lietorch
Successfully built MAST3R-SLAM
Failed to build lietorch
ERROR: Failed to build installable wheels for some pyproject.toml based projects (lietorch)
(mast3r502) juhwan@DESKTOPJUHWAN:~/MASt3R-SLAM$

=========================================================================

PR5GRAMM2R, 2025-04-30:
Actually, I found lietorch which hectorpiteau modified about two hours ago. ( Link : https://github.com/hectorpiteau/lietorch )

I modified file pyproject.tomi

this part : "lietorch @ git+https://github.com/princeton-vl/lietorch.git",

to this part : "lietorch @ git+https://github.com/hectorpiteau/lietorch.git", 

Then I installed mast3r-slam again with this commend " pip install --no-build-isolation -e . ",

and I installed pytorch 2.8.0 again because with this part

 "lietorch @ git+https://github.com/hectorpiteau/lietorch.git", 

pytorch 2.6.0 is reinstalled automatically.

When I installing pytorch 2.8.0 , the error occured but I neglected it.

==============================================================================

ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
lietorch 0.3 requires torch==2.6.0, but you have torch 2.8.0.dev20250429+cu128 which is incompatible.
Successfully installed nvidia-cublas-cu12-12.8.3.14 nvidia-cuda-cupti-cu12-12.8.57 nvidia-cuda-nvrtc-cu12-12.8.61 nvidia-cuda-runtime-cu12-12.8.57 nvidia-cudnn-cu12-9.8.0.87 nvidia-cufft-cu12-11.3.3.41 nvidia-curand-cu12-10.3.9.55 nvidia-cusolver-cu12-11.7.2.55 nvidia-cusparse-cu12-12.5.7.53 nvidia-cusparselt-cu12-0.6.3 nvidia-nccl-cu12-2.26.2 nvidia-nvjitlink-cu12-12.8.61 nvidia-nvtx-cu12-12.8.55 sympy-1.13.3 torch-2.8.0.dev20250429+cu128

==============================================================================

Anyway,  all the things installed, so I tested the TUM dataset. 

Then, it worked with no error!

With RTX 5070 ti, TUM dataset evaluation took about 10 minutes.

I was really appreciated, because this works caused me lots of time.

And I want to appreciate to you for your assistance, sir.

I will test your new suggestion with https://github.com/nicogorlo/lietorch .

Thanks again! ^0^

closed 2025-05-22 by PR5GRAMM2R

IgnorAnsel, 2025-10-30:
> Hi,
> 
> Try the changes I am proposing in [#86](https://github.com/rmurai0610/MASt3R-SLAM/pull/86).
> 
> Cheers.

... loading model from checkpoints/MASt3R_ViTLarge_BaseDecoder_512_catmlpdpt_metric.pth
2.8.0+cu128
True
12.8
Warning, cannot find cuda-compiled version of RoPE2D, using a slow pytorch version instead
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/ansel/anaconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/spawn.py", line 122, in spawn_main
    exitcode = _main(fd, parent_sentinel)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ansel/anaconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/spawn.py", line 132, in _main
    self = reduction.pickle.load(from_parent)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ansel/anaconda3/envs/mast3r-slam/lib/python3.11/site-packages/torch/multiprocessing/reductions.py", line 181, in rebuild_cuda_tensor
    storage = storage_cls._new_shared_cuda(
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ansel/anaconda3/envs/mast3r-slam/lib/python3.11/site-packages/torch/storage.py", line 1457, in _new_shared_cuda
    return torch.UntypedStorage._new_shared_cuda(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
torch.AcceleratorError: CUDA error: invalid resource handle
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

instantiating : AsymmetricMASt3R(enc_depth=24, dec_depth=12, enc_embed_dim=1024, dec_embed_dim=768, enc_num_heads=16, dec_num_heads=12, pos_embed='RoPE100',img_size=(512, 512), head_type='catmlp+dpt', output_mode='pts3d+desc24', depth_mode=('exp', -inf, inf), conf_mode=('exp', 1, inf), patch_embed_cls='PatchEmbedDust3R', two_confs=True, desc_conf_mode=('exp', 0, inf), landscape_only=False)
<All keys matched successfully>
2.8.0+cu128
True
12.8
Warning, cannot find cuda-compiled version of RoPE2D, using a slow pytorch version instead
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/ansel/anaconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/spawn.py", line 122, in spawn_main
    exitcode = _main(fd, parent_sentinel)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ansel/anaconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/spawn.py", line 132, in _main
    self = reduction.pickle.load(from_parent)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ansel/anaconda3/envs/mast3r-slam/lib/python3.11/site-packages/torch/multiprocessing/reductions.py", line 181, in rebuild_cuda_tensor
    storage = storage_cls._new_shared_cuda(
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ansel/anaconda3/envs/mast3r-slam/lib/python3.11/site-packages/torch/storage.py", line 1457, in _new_shared_cuda
    return torch.UntypedStorage._new_shared_cuda(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
torch.AcceleratorError: CUDA error: invalid resource handle
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

^CTraceback (most recent call last):
  File "/home/ansel/works/MASt3R-SLAM/main.py", line 229, in <module>
    backend.start()
  File "/home/ansel/anaconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/process.py", line 121, in start
    self._popen = self._Popen(self)
                  ^^^^^^^^^^^^^^^^^
  File "/home/ansel/anaconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/context.py", line 224, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ansel/anaconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/context.py", line 288, in _Popen
    return Popen(process_obj)
           ^^^^^^^^^^^^^^^^^^
  File "/home/ansel/anaconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/popen_spawn_posix.py", line 32, in __init__
    super().__init__(process_obj)
  File "/home/ansel/anaconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/popen_fork.py", line 19, in __init__
    self._launch(process_obj)
  File "/home/ansel/anaconda3/envs/mast3r-slam/lib/python3.11/multiprocessing/popen_spawn_posix.py", line 62, in _launch
    f.write(fp.getbuffer())
KeyboardInterrupt
[W1030 14:08:55.330710990 CudaIPCTypes.cpp:100] Producer process tried to deallocate over 1000 memory blocks referred by consumer processes. Deallocation might be significantly slowed down. We assume it will never going to be the case, but if it is, please file but to https://github.com/pytorch/pytorch

Excuse me?

ZerunLiu, 2026-01-09:
> 这是由于 lietorch 包中存在的相同问题导致的。 我已经向他们提交了一个 PR。 目前，您可以使用我 fork 的 lietorch 包：
> 
> https://github.com/nicogorlo/lietorch
> 
> 如果有效请告诉我！

I have tried this link "https://github.com/nicogorlo/lietorch", finally it works, thank you for help.  And my GPU is RTX 5880, CUDA version is 12.8. I followed your instructions and solved many problems