rmurai0610/MASt3R-SLAM #81: Evaluation on ScanNet

VladimirYugay, 2025-04-21:
Hey there. Congrats on the fantastic work!

I'm trying to evaluate the tracking accuracy on the ScanNet videos. However, the results seem slightly off to me. First, I run the method in uncalibrated mode on every ScanNet video with a stride of 10 to avoid out-of-memory errors. 

Then I load the poses produced by the method:
```
def load_tum_trajectory(filepath):
    data = np.loadtxt(filepath)
    if len(data.shape) == 1:  # too short of a video
        return None, None
    timestamps = data[:, 0]
    positions = data[:, 1:4]                     # (N, 3)
    quats_xyzw = data[:, 4:8]                   # qx, qy, qz, qw

    rotations = Rotation.from_quat(quats_xyzw)          # (N,) Rotation objects
    matrices = rotations.as_matrix()             # (N, 3, 3)

    poses = np.eye(4)[None].repeat(len(data), axis=0)  # (N, 4, 4)
    poses[:, :3, :3] = matrices
    poses[:, :3, 3] = positions
    return timestamps, poses
```

Load the GT trajectory:
```
def load_scannet_poses(scene_dir, timestamps):

    scene_dir = Path(scene_dir)
    video_path = scene_dir / "video.mp4"
    pose_path = scene_dir / "poses.npz"

    vr = VideoDecoder(str(video_path))
    fps = vr.metadata.average_fps
    total_frames = vr.metadata.num_frames

    c2ws = np.load(pose_path)["poses"]
    # Take everything until first invalid pose
    inf_ids = np.where(np.isinf(c2ws).any(axis=(1, 2)))[0]
    if inf_ids.size > 0:
        c2ws = c2ws[:inf_ids.min()]

    # Convert to world coordinate frame
    w2cs = np.linalg.inv(c2ws)

    # Move to the origin for visualization
    w2cs = np.linalg.inv(w2cs[0]) @ w2cs
    
    timestamp_to_w2c = {}
    selected_poses = []
    for i in range(c2ws.shape[0]):
        cur_ts = i / fps
        timestamp_to_w2c[cur_ts] = w2cs[i]
    
    selected_poses = []
    for ts in timestamps:
        if ts in timestamp_to_w2c:
            selected_poses.append(timestamp_to_w2c[ts])
    
    return np.array(selected_poses)  
```

I use the following evo wrapper to evaluate the pose:

```
def eval_trajectory(poses_est, poses_gt, frame_ids, align=False, delta=1):

    traj_ref = PoseTrajectory3D(
        positions_xyz=poses_gt[:, :3, 3],
        orientations_quat_wxyz=Rotation.from_matrix(poses_gt[:, :3, :3]).as_quat(scalar_first=True),
        timestamps=frame_ids)
    traj_est = PoseTrajectory3D(
        positions_xyz=poses_est[:, :3, 3],
        orientations_quat_wxyz=Rotation.from_matrix(poses_est[:, :3, :3]).as_quat(scalar_first=True),
        timestamps=frame_ids)

    ate_result = main_ape.ape(
        deepcopy(traj_ref),
        deepcopy(traj_est),
        est_name="traj",
        pose_relation=PoseRelation.translation_part,
        align=align,
        correct_scale=align)
    ate = ate_result.stats["rmse"]

    are_result = main_ape.ape(
        deepcopy(traj_ref),
        deepcopy(traj_est),
        est_name="traj",
        pose_relation=PoseRelation.rotation_angle_deg,
        align=align,
        correct_scale=align)
    are = are_result.stats["rmse"]

    # RPE rotation and translation
    rpe_rots_result = main_rpe.rpe(
        deepcopy(traj_ref),
        deepcopy(traj_est),
        est_name="traj",
        pose_relation=PoseRelation.rotation_angle_deg,
        align=align,
        correct_scale=align,
        delta=delta,
        delta_unit=Unit.frames,
        rel_delta_tol=0.01,
        all_pairs=True)
    rpe_rot = rpe_rots_result.stats["rmse"]

    rpe_transs_result = main_rpe.rpe(
        deepcopy(traj_ref),
        deepcopy(traj_est),
        est_name="traj",
        pose_relation=PoseRelation.translation_part,
        align=align,
        correct_scale=align,
        delta=delta,
        delta_unit=Unit.frames,
        rel_delta_tol=0.01,
        all_pairs=True)
    rpe_trans = rpe_transs_result.stats["rmse"]

    plot_mode = PlotMode.xz
    fig = plt.figure()
    ax = evo.tools.plot.prepare_axis(fig, plot_mode)
    ax.set_title(f"ATE: {round(ate, 3)}, ARE: {round(are, 3)}")
    evo.tools.plot.traj(ax, plot_mode, traj_ref, "--", "gray", "gt")
    evo.tools.plot.traj_colormap(
        ax,
        traj_est,
        ate_result.np_arrays["error_array"],
        plot_mode,
        min_map=ate_result.stats["min"],
        max_map=ate_result.stats["max"],
    )
    ax.legend()

    buffer = io.BytesIO()
    plt.savefig(buffer, format='png', dpi=90)
    buffer.seek(0)

    pillow_image = Image.open(buffer)
    pillow_image.load()
    buffer.close()
    plt.close(fig)

    return {
        "ate": ate,
        "are": are,
        "rpe_rot": rpe_rot,
        "rpe_trans": rpe_trans
    }, pillow_image

```

This gives me some weird results. For example, here's the result for the sequence `scene0794_00` without alignment:

[image]
 

Is this an expected behaviour, or am I missing something?

edexheim, 2025-04-22:
Hi,

Thanks for the question.  Do you know how many keyframes were produced?

I'll try to set up a dataloader for ScanNet to see.  Is your data in mp4 format? I have this sequence but it seems to be a set of 1113 images, so also a little surprised that it goes out of memory.

One other potential issue is that ScanNet images often have an invalid border of pixels (I think from registering RGB and depth) so will likely need to crop those out.

Best,
Eric

VladimirYugay, 2025-04-23:
Thanks for such a quick reply. 

Yes, there are 30 keyframes. I converted the scans to videos to reduce the number of files for the cluster. Here it is:

https://github.com/user-attachments/assets/2893449d-de8d-4012-8fc3-21efd6ead819. 

I ran an evaluation on the entire ScanNet test set, and it ran out of memory only for some long videos (~4,000 frames). I tried increasing `SharedKeyframes(manager, h, w, 4000)`, but 4k is the maximum I can afford. Setting `subsample: 10` in `base.yaml` together with a 4K buffer helps overcome OOM.

VladimirYugay, 2025-04-29:
Hey there. There was an issue on my end during the evaluation. I didn't take stride into account, which I set to 10 in the default configuration, when computing the timestamps for the ground-truth poses. Additionally, I inverted the ground-truth camera poses, assuming that the predicted poses are in the world-to-camera coordinate system.

closed 2025-04-29 by VladimirYugay