rmurai0610/MASt3R-SLAM #82: pip install -e thirdparty/mast3r Error

Heavenchaos, 2025-04-23:
Hello! I met an error when doing this command. It shows:

*ModuleNotFoundError: No module named 'torch'

seems i do not have torch. but problem is, i installed torch already in conda. 
i tried pip install torch, but shows: *Requirements already satisfied!

using windows11, powershell.
Thanks for any help!

alexxxroz, 2025-04-24:
> I would suggest to move to either Ubuntu/Linux or WSL2 (I used Ubuntu 22.04). And hopefully, this issue will be resolved.

Weirdly, I'm using Ubuntu 24 and facing the same issue...

theoden8, 2025-04-24:
use torch 2.3.1 (stable), worked for me

rmurai0610, 2025-05-02:
Hi, can you try `pip install --no-build-isolation -e thirdparty/mast3r`
I don't have powershell or Ubuntu 24 setup but if the issue is with missing torch it might help

closed 2025-06-09 by rmurai0610