rmurai0610/MASt3R-SLAM #83: pip install -e thirdparty/in3d

darsh1406, 2025-04-23:
When I run this command I am getting this error:

(slam-env) root@612d8fc49828:/workspace/MASt3R-SLAM# pip install -e thirdparty/in3d
Obtaining file:///workspace/MASt3R-SLAM/thirdparty/in3d
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Processing ./thirdparty/in3d/thirdparty/pyimgui
ERROR: imgui@ file:///workspace/MASt3R-SLAM/thirdparty/in3d/thirdparty/pyimgui from file:///workspace/MASt3R-SLAM/thirdparty/in3d/thirdparty/pyimgui (from in3d==0.0.1) does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.

theoden8, 2025-04-25:
i fixed this by

```
uv venv
. .venv/bin/activate
cd thirdparty/in3d
uv pip install -e .
```

rmurai0610, 2025-05-02:
Hi, did you run `git submodule update --init --recursive`?
The error seems to hint that pyimgui/imgui directory is empty.

closed 2025-06-09 by rmurai0610