rmurai0610/MASt3R-SLAM #84: Evaluation on Reconstruction

Sissi-lu, 2025-04-24:
Hi, 

Thank you for your excellent work in combining the MASt3R into the SLAM. I tried to reproduce the paper's results and use this model to see whether it is acceptable in a medical SLAM scenario. And I found some questions as follows:

1、I did not find the evaluation metrics for the reconstruction result (i.e., accuracy, completion, and chamber). I also tried to write the eval result, but the GT point cloud is too large if I generate it from the depth corresponding to the RGB images (7-scenes dataset). Could you publish this part of the code or give the instructions?

2、I noticed that not all the folders contain point clouds (ply) in EuRoC. So, did you evaluate only the scenes with point clouds?

3、I found it suitable for doing a slam in a gastrointestinal tract dataset. However, MASt3R did not see this kind of data. Why can it achieve this?

Many thanks for considering my request.

rmurai0610, 2025-05-02:
Hi, we use only the depth from frames which were registered as keyframe to create the GT pointcloud. 
Similarly for EuRoC, we prune out the regions which are not observed by any of the keyframes.
We evaluate only on the EuRoC Vicon room, and all of the vicon rooms sequences has pointclouds.
I am not sure about the gastrointestinal tract dataset. MASt3R seems to generalise well beyond training data (e.g. it can more or less reconstruction static human, even though it's never seen one in the dataset) so I assume something similar is happening here. 

Hope these help!

closed 2025-05-02 by rmurai0610

Sissi-lu, 2025-05-07:
Thank you once again for answering!

Sissi-lu, 2025-05-10:
Hi,

Sorry to bother you once again. I still have some problems with reconstruction evaluation.

For reproducing reconstruction evaluation on 7 scenes, I followed your suggestion to use the depth from frames that were registered as keyframes to create the GT point cloud. And I found that the results using MASt3R-SLAM are higher than in the papers, as shown below:

[image]

This is the evaluation code I wrote, as I did not find the reconstruction evaluation code in your repository:

`import argparse
import os.path
from natsort import natsorted
import yaml
import numpy as np
import open3d as o3d
import cv2
import torch
from mast3r_slam.dataloader import Intrinsics

def quaternion_to_rotation_matrix(qx, qy, qz, qw):
    """
    Convert a quaternion (qx, qy, qz, qw) to a 3x3 rotation matrix.
    Assumes quaternion is normalized (qx^2 + qy^2 + qz^2 + qw^2 = 1).

    Args:
        qx, qy, qz: Vector components of the quaternion.
        qw: Scalar component of the quaternion.

    Returns:
        R: 3x3 rotation matrix (numpy array).
    """
    # Ensure the quaternion is normalized
    norm = np.sqrt(qx ** 2 + qy ** 2 + qz ** 2 + qw ** 2)
    if not np.isclose(norm, 1.0, atol=1e-6):
        qx, qy, qz, qw = qx / norm, qy / norm, qz / norm, qw / norm

    # Compute the rotation matrix
    R = np.zeros((3, 3))
    R[0, 0] = 1 - 2 * qy ** 2 - 2 * qz ** 2
    R[0, 1] = 2 * qx * qy - 2 * qz * qw
    R[0, 2] = 2 * qx * qz + 2 * qy * qw
    R[1, 0] = 2 * qx * qy + 2 * qz * qw
    R[1, 1] = 1 - 2 * qx ** 2 - 2 * qz ** 2
    R[1, 2] = 2 * qy * qz - 2 * qx * qw
    R[2, 0] = 2 * qx * qz - 2 * qy * qw
    R[2, 1] = 2 * qy * qz + 2 * qx * qw
    R[2, 2] = 1 - 2 * qx ** 2 - 2 * qy ** 2

    return R

def pose_to_transformation_matrix(quat):
    """
    Convert a pose (tx, ty, tz, qx, qy, qz, qw) to a 4x4 transformation matrix.

    Args:
        tx, ty, tz: Translation components.
        qx, qy, qz, qw: Quaternion components (qw is the scalar part).

    Returns:
        T: 4x4 homogeneous transformation matrix (numpy array).
    """

    tx, ty, tz, qx, qy, qz, qw = quat
    # Get the 3x3 rotation matrix from the quaternion
    R = quaternion_to_rotation_matrix(qx, qy, qz, qw)

    # Create the 4x4 transformation matrix
    T = np.eye(4)  # Initialize 4x4 identity matrix
    T[0:3, 0:3] = R  # Set the top-left 3x3 block to the rotation matrix
[... 245 lines cut ...]

    # Convert depth images to point clouds
    # gt_points = []
    # for depth_img, pose, rgb_img in zip(depth_images, poses, rgb_images):
    #     pcd = depth_to_point_cloud(depth_img, intrinsics, pose, rgb_img=rgb_img)
    #     gt_points.append(pcd)

    gt_pcds = depth_to_point_cloud(depth_images, intrinsics, poses, rbg_images=rgb_images)

    ply_path = os.path.join(base_path, "logs/%s/%s/%s/%s.ply"%(datatype, wo_calib, data_name, data_name))
    assert os.path.exists(ply_path), "there is no path: %s"%(ply_path)
    pred_pcd = o3d.io.read_point_cloud(ply_path)

    o3d.visualization.draw_geometries([pred_pcd], window_name="Predicted Point Cloud")

    o3d.visualization.draw_geometries([gt_pcds], window_name="GT Point Cloud")

    # Evaluate
    metrics = evaluate_reconstruction(pred_pcd, gt_pcds, threshold=0.5)
    print("\nFinal Evaluation Metrics:")
    print(f"Accuracy (RMSE): {metrics['Accuracy_RMSE']:.4f}")
    print(f"Completion (RMSE): {metrics['Completion_RMSE']:.4f}")
    print(f"Chamfer Distance: {metrics['Chamfer_Distance']:.4f}")`

Is the code right or not? If it is somewhere wrong, please point it out. Thank you once again for considering my request!