rmurai0610/MASt3R-SLAM #90: Is it normal for the cloud to have many layers on the walls and floors?

mariodiazgomez, 2025-05-08:
Hello, first of all, congratulations on the excellent work. It's truly incredible to be able to obtain such information from a common video. On the other hand, I would like to know if there is any way to improve the quality of the cloud. I am currently using it in WSL on W11 with an NVIDIA RTX 4060 ti 16GB, i9-12900k, and 48GB of RAM. The video with which this cloud was generated was recorded with the iPhone 13's ultra-wide-angle camera. I have processed it in both 720p and 1080p and have obtained similar results. I would like to know what could be causing the cloud to look layered. I should add that I am not using a specific calibration for the camera, only this command:

`python main.py --dataset video.mp4 --config config/base.yaml`

Any suggestions for improving this?

[image]

[image]

[image]

lAi27088, 2025-05-08:
Hi, I'm trying to install MASt3R-SLAM on WSL2 but I'm running into an `undefined symbol` error when importing `mast3r_slam_backends.so`.

Could you kindly share your setup when you successfully installed it?
Specifically:
- Ubuntu / WSL2 version?
- CUDA toolkit version (`nvcc --version`)?
- NVIDIA driver and supported CUDA (`nvidia-smi`)?
- PyTorch version (`torch.__version__`) and installed CUDA?
- Which branch did you install from? (`main` or `windows`?)
- Did you manually install any other dependencies (e.g. g++, torch-nightly, patched setup)?

I'd really appreciate any insights. Thanks so much!

LimHyungTae, 2025-05-11:
I'm not the author of MAST3R-SLAM, but my gut feeling is that there seems to be no reliable and repeatable ones in your video.