rmurai0610/MASt3R-SLAM #91: How to skip feature points?

laiyuzhi, 2025-05-13:
Hi @rmurai0610 , I have same question as #48 . Similarly I feel that it should be done without considering the correspondence of the masked feature points. But for example, should I set the D of these points to 0 or something else? Can you tell me in a little more detail how to skip these feature points?

ppsohandsome, 2025-05-13:
> Hi @rmurai0610 , I have same question as #48 . Similarly I feel that it should be done without considering the correspondence of the masked feature points. But for example, should I set the Q of these points to 0 or something else? Can you tell me in a little more detail how to skip these feature points?      

i don't know if there are easier way,but for me i add a alpha channel，which means turn RGB into RGBA. In that way, you should change one class in dust3r,making it support 4 channel frame,and making feather matching process ignore points of alpha channel

ppsohandsome, 2025-05-13:
> Did you train the model with rgba or how did you get the featuring to skip certain masked points?
> 
> ps: Do u have Wechat?  (lyz903190021)

包的兄弟，加你