rmurai0610/MASt3R-SLAM #108: Where to find the 3d Point cloud video

aniket-professional2025, 2025-08-12:
Ihave run the code: python main.py --dataset <path/to/video>.mp4 --config config/base.yaml on my video.mp4 file. The code runs succesfully with the statement done. 

However, I am not sure where the 3d point cloud object is stored. I am unable to find it. 

Need help to know where the output is stored

Daniel00008, 2025-09-07:
I have the same problem. Have you found the files for visualisation?

aniket-professional2025, 2025-09-08:
Yes. As per I remember, they will be in the **logs** folder (as a .ply file). However, it will give the .ply file. You need to use the SpatialLM to convert the .ply file into an .rrd file and use rerun to visualize it. It will not give the video attached. It will just show the layout of the objects/points you have taken.